stack-of-tasks/pinocchio #1377: API that fully update the robot's data

junhyeokahn, 2021-01-21:
Hello,
First of all, I appreciate your effort to maintain this nice repo.

I have a quick question about how Pinocchio updates the robot's properties.
It looks like Pinocchio has multiple steps to update the robots quantities given ```q, qdot```.
For instance, ```forwardKinematics``` does not update the frames' placement, and therefore ```updateFramePlacements``` should be called afterward.
Also, ```ccrba``` separately update centroidal quantities.

I am wondering is there any function where I could update all the robot's quantities at once and save results at ```Data```class?
If not, could you kindly explain what kind of APIs should be called in which order to correctly update the quantities?

jcarpent, 2021-01-21:
There is the function computeAllTerms that does everything, except centroidal quantities. I will update it accordingly.

gabrielebndn, 2021-01-21:
> There is the function computeAllTerms that does everything, except centroidal quantities

Since @junhyeokahn mentioned frames, I will mention that `computeAllTerms` does not update frame placements either

BTW, of you just want to call `forwardKinematics(model, data, q);` and `updateFramePlacements(model, data);` all in one call and nothing else, there's `framesForwardKinematics(model, data, q);` doing it

jcarpent, 2021-01-21:
Indeed, but it is easy to call `updateFramePlacements(model, data)` after calling `computeAllTerms` without additional burden. At this stage, there is no need to add `updateFramePlacements` in `computeAllTerms`

gabrielebndn, 2021-01-21:
> Indeed, but it is easy to call `updateFramePlacements(model, data)` after calling `computeAllTerms` without additional burden. At this stage, there is no need to add `updateFramePlacements` in `computeAllTerms`

There isn't of course, I was just said it so that it would be clear to @junhyeokahn since he explicitly mentioned this

junhyeokahn, 2021-01-21:
@jcarpent @gabrielebndn Thanks for the reply.
So, after #1378 incorporated,  I could probably call ```computeAllTerms``` and then ```updateFramePlacements```, which will update all quantities.

Then, I might be able to access (correct me if I am wrong)
- centroidal quantities from ```data.hg```, ```data.Ag```, ```data.Ig``` without explicitly calling ```ccrba```,
- com quantities from ```data.com[0]```, ```data.vcom[0]``` without explicitly calling ```centerOfMass```,
- com jacobian from ```data.Jcom``` instead of calling jacobianCenterOfMass```,
- nonlinear term from ```data.nle``` instead of calling ```nonLinearEffects```,
- gravity term from ```data.g``` instead of calling ```computeGeneralizedGraivity```,
- mass matrix from ```data.M``` instead of calling ```crba```,
- frame placement from ```data.oMf``` instead of calling ```updateFramePlacement```.

Follow up question is then,
is there any way I can access the following quantities from ```data``` as well
- frame velocities in world frame (velocity version of ```data.oMf```),
- frame jacobian in world frame (so ```frame velocities = jacobian * qodt```),
- frame jacobian time derivative in world frame (so that ```frame acc = jacobian_dot * qdot + jacobian*qddot```),
- centroidal map time derivative in world frame (so that its top 3 quantity can represent ```com jacobian dot * total_robot_mass```?

I saw in the ```robot_wrapper``` these quantities are achieved from
- ```getFrameVelocity(model, data, frame_id)```,
- ```getFrameJacobian(model, data, frame_id, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)```,
- ```getFrameJacobianTimeVariation(model, data, frame_id, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)```,
- ```computeCentroidalMapTimeVariation(model, data, q, qdot)```,
but just wondering if I could access these quantities from ```data``` I as do for the other quantities.

One quick last question is how quickly conda (python bindings) reflects the changes in the master branch?

gabrielebndn, 2021-01-21:
> I saw in the `robot_wrapper` these quantities are achieved from
>
>     * `getFrameVelocity(model, data, frame_id)`,
>
>     * `getFrameJacobian(model, data, frame_id, [...]

Those are exactly the functions you need. Frame velocities, accelerations and Jacobians are not stored anywhere in `data` but they are computed on the fly (the same is true for joint Jacobians BTW).

The result of `computeCentroidalMapTimeVariation` is stored is `data.dAg` (you can check the [online doc](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/) for that) and it is also computed by `dccorba` (the two algorithms seem to be equivalent, at least in terms of output)

jcarpent, 2021-01-21:
> Then, I might be able to access (correct me if I am wrong)
> 
>     * centroidal quantities from `data.hg`, `data.Ag`, `data.Ig` without explicitly calling `ccrba`, YES
> 
>     * com quantities from `data.com[0]`, `data.vcom[0]` without explicitly calling `centerOfMass`, YES
> 
>     * com jacobian from `data.Jcom` instead of calling jacobianCenterOfMass```, YES
> 
>     * nonlinear term from `data.nle` instead of calling `nonLinearEffects`, YES
> 
>     * gravity term from `data.g` instead of calling `computeGeneralizedGraivity`, YES
> 
>     * mass matrix from `data.M` instead of calling `crba`, YES
> 
>     * frame placement from `data.oMf` instead of calling `updateFramePlacement`. NO!!!

You should still call `updateFramePlacements(model,data)` which pretty much easy.

jcarpent, 2021-01-21:
The question is why do you need dJ/dt or dAg/dt?

jcarpent, 2021-01-21:
> One quick last question is how quickly conda (python bindings) reflects the changes in the master branch?

For each new release, conda is compiling and making it available to anyone on conda-forge. What is your timeline?
In any cases, it is very easy to install Pinocchio from sources under a conda environment.

junhyeokahn, 2021-01-21:
@gabrielebndn @jcarpent Thanks for clarifying all the questions
It resolves all of my questions!

> What is your timeline?

I don't have a strict timeline but I was just wondering.

I use ```dJ/dt``` to control operational space task in acceleration level (using the equation ```xddot = dJ/dT * qdot + J * qddot```).

jcarpent, 2021-01-21:
This is a drift term, so you don't need to compute this term at all. Assuming qddot == 0, you get your quantity for free from forward kinematics. 
Just call `computeAllTerms`, it is all what you need!

jcarpent, 2021-01-22:
Solved by #1378.

closed 2021-01-22 by jcarpent