stack-of-tasks/pinocchio #1381: kinematics quantities regarding linkage com frame

junhyeokahn, 2021-01-24:
Hello,
I would like to get kinematics quantities at linkage com frame.
I thought I could use an API such as ```pin.updateFramePlacement```, ```pin.getFrameJacobian```, and ```pin.getFrameVelocity```, but I couldn't get the right values using these.
I made a simple three-linkage manipulator example and tried to achieve placement, jacobian, and velocity at both joint frame and link com frame (I am attaching a code and the URDF file I used).
It looks like quantities at joint frame are printed out as I expected but the ones at link com frame are not.

Am I using the right functions to get kinematics quantities at the joint frame and the link com frame?

```
import os                                                                                                                                                                                                                                                                                                                
import sys                                                                                                                                                                                                                                                                                                               
cwd = os.getcwd()                                                                                                                                                                                                                                                                                                        
sys.path.append(cwd)                                                                                                                                                                                                                                                                                                     
                                                                                                                                                                                                                                                                                                                         
import pinocchio as pin                                                                                                                                                                                                                                                                                                  
import numpy as np                                                                                                                                                                                                                                                                                                       
                                                                                                                                                                                                                                                                                                                         
urdf_file = cwd + "/three_link_manipulator.urdf"                                                                                                                                                                                                                                                                         
model = pin.buildModelFromUrdf(urdf_file)                                                                                                                                                                                                                                                                                
data = model.createData()                                                                                                                                                                                                                                                                                                
print(model)                                                                                                                                                                                                                                                                                                             
                                                                                                                                                                                                                                                                                                                         
q = np.zeros(3)                                                                                                                                                                                                                                                                                                          
qdot = np.ones(3)                                                                                                                                                                                                                                                                                                        
                                                                                                                                                                                                                                                                                                                         
pin.forwardKinematics(model, data, q, qdot)                                                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                                                                                                         
## Print Frame Names                                                                                                                                                                                                                                                                                                     
print([frame.name for frame in model.frames])                                                                                                                                                                                                                                                                            
                                                                                                                                                                                                                                                                                                                         
## Calculate j2 placement                                                                                                                                                                                                                                                                                                
j2_frame = model.getFrameId('j2')                                                                                                                                                                                                                                                                                        
j2_translation = pin.updateFramePlacement(model, data, j2_frame)                                                                                                                                                                                                                                                         
print("j2 translation")                                                                                                                                                                                                                                                                                        [... cut ...]                                                                                      
print("l2 vel")                                                                                                                                                                                                                                                                                                          
print(l2_vel)  
```
[files.zip](https://github.com/stack-of-tasks/pinocchio/files/5861380/files.zip)

jcarpent, 2021-01-24:
You have forgotten the optional argument of `getFrameVelocity` which is related to the reference Frame info.

junhyeokahn, 2021-01-24:
@jcarpent Thanks for the quick reply.
I agree I forgot the argument for the reference Frame.
My question here was how can I get velocity, jacobian, placement at the linkage COM frame rather than at the joint frame.
For instance, I expect ```l2_translation``` should have ```p=[1.5, 0, 0]``` and ```j2_translation``` should have ```p=[1, 0, 0]``` but the code returns the same value which is ```p=[1, 0, 0]```.

jcarpent, 2021-01-24:
You should explicitly add a Frame related to each centre of mass, and then use the classic operations.
The position of the centre of mass in the joint Frame is given by `model.inertias[joint_id].lever()`.

junhyeokahn, 2021-01-24:
Oh I see,
I would create com frames in URDF with a fixed joint.

Thanks!

closed 2021-01-24 by junhyeokahn