stack-of-tasks/pinocchio #1382: differentiation for a unit vector in the world frame

ddliu365, 2021-01-24:
Hi,

I want to differentiate a unit vector in the world frame as in this equation
[image]
where, R_{We}(q) is the rotation matrix of end effector in the world frame and v_{ey} is a unit vector in the local frame of end effector.
the rotation matrix is 3x3 dimension, and v_{ey} is3x1 dimension, so l is 3x1 dimension. 
For the expected differentiation output, the Jacobian should be 3xnq dimension.

I plan to use `pinocchio.getFrameJacobian(model, data, self.Rref.id,
                                                    pinocchio.ReferenceFrame.WORLD)[3:, :]` to represent the Jacobian of orientation part(3xnq), but if we apply the constant vector v_{ey} on the orientation Jacobian which is 3x1, it seems the output of differentiation is 1xnq.

I think I made a mistake for the derivation. Do you know how to address this issue?

jcarpent, 2021-01-24:
The differentiation operator on SO(3) is the cross product: R^t * dR/dt = w x
You should then get the Jacobian of l as follows: `dl/dq = J x l` where the cross operator is defined column-wise.

closed 2021-01-24 by jcarpent

ddliu365, 2021-01-25:
sorry, I didn't get why it is `dl/dq = J x l` instead of `dl/dq = J x v_{ey}` since `v_{ey}` is a constant vector here
Would you mind to give more derivation

ddliu365, 2021-01-25:
for the cross operator here, you mean I should do cross product for l and each column vector in J?

jcarpent, 2021-01-25:
l is the expression of v_ey in the world coordinate system.

> for the cross operator here, you mean I should do cross product for l and each column vector in J?

Yes.

By the way, I hope you won't forget to cite/acknowledge Pinocchio when publishing your work.

ddliu365, 2021-01-25:
sure, I will. I think I need to make up some geometries stuff, do you have any recommended material for understanding this derivation ? right now I am reading the rigid body dynamics algorithms by Featherstone

jcarpent, 2021-01-25:
A Mathematical Introduction to Robotic Manipulation, Murray et al. is of course another very good entry point.

ddliu365, 2021-01-26:
I just quickly went through the content.
Based on my understanding, this Jocobian is not the analytic Jacobian w.r.t to `q`. My previous notation is totally wrong.
It is a geometric Jacobian, I think. 
For a frame translation Jacobian, we have
[image]
For a frame orientation, we have 
[image]
so, we are basically finding something which connects the derivates of our vector and the joint velocity. Based on this understanding, we have the Jacobian in the following bracket to represent the "derivative w.r.t `q`"
[image]
Does I understand correct?

ddliu365, 2021-01-27:
Another question is: is there a second order differentiation for `l` w.r.t `q`

jcarpent, 2021-01-27:
> I just quickly went through the content you recommended.
> Based on my understanding, this Jocobian is not the analytic Jacobian w.r.t to `q`. My previous notation is totally wrong.
> It is a geometric Jacobian, I think.

I don’t get what you mean. This is not clear.

jcarpent, 2021-01-27:
> Do I understand correct?

This is exactly what I’ve written before.

jcarpent, 2021-01-27:
> Another question is: is there a second order differentiation for `l` w.r.t `q`
> if there is, should it be like the following?
> [image]

Your derivations are not correct. Did you check twice by yourself?

ddliu365, 2021-01-27:
Sorry for my unclear question. Thanks for your remind my problem and I will notice this issue in the future question. 

Please ignore my understanding about geometry Jacobian.

From the derivation, I understanding why it is `J x l` instead of `J x v_{ey}`. Your explanation is absolutely right.

I am working on a `crocoddyl` problem like you mentioned. The cost function needs a first order differentiation as well as a second order differentiation w.r.t `q`. I got struck by the dimension issue in the second order differentiation. The previous description is only a part of the problem. here is the full cost function and the first order differentiation based on what you showed to me

[image]

Based on `crocoddyl` [convention](https://github.com/loco-3d/crocoddyl/blob/b0eeaa5713166d7e6955454b90aedf0fc940baa1/include/crocoddyl/multibody/costs/frame-velocity.hxx#L82), the second order differentiation (i.e. Hessian )is expected to be approximated by 
[image]
But I am not sure if this is correct for my problem. 
If I ignore this convention,  I derived the second order like this:
[image]
here, `J_O` and `J_p` are the spatial linear Jacobian and spatial angular Jacobian. `v_{Wg}` is the end effector velocity in the inertial frame. `v_{gy}` is a vector in the body frame of end effector. `R_{Wg}` is the rotation matrix map frame of end effector `g` to inertial frame `W`. `l` is the cost function, `l_q` is the first order differentiation of `l ` w.r.t. `q`, and `l_{qq}` is the second order differentiation of `l` w.r.t  `q`. 
This derivation I ignored the derivative of this part(marked with red oval shape), but I am not sure if this is right:
[image]

jcarpent, 2021-01-27:
@ddliu365 Did you completely read my message?

> PS: what you are asking here is some supervisor's feedback. It is hard at the project level to help everybody on his/her research issues, as we have other obligations which prevent us to be as responsive as you would like. We are already doing our best to provide you with some scientific guidance, either on Pinocchio or Crocoddyl, but you are very demanding. I would suggest asking further helps from your research institution.

ddliu365, 2021-01-27:
sorry, I missed that part.

ddliu365, 2021-01-27:
Please try to response when you are free. Thank you very much.