stack-of-tasks/pinocchio #1384: Redundant frame placement update in getFrameJacobian()

nuft, 2021-01-27:
The method `pinocchio::getFrameJacobian()` updates the frame placement even though the documentation tells to call `pinocchio::framesForwardKinematics()` beforehand. I believe this frame placement update is unnecessary and can be removed. It would also allow taking data as a const reference, which would be more consistent with the computeX() and getX() pattern.

The same could be done `pinocchio::getFrameJacobianTimeVariation()`.

What do you think?

jcarpent, 2021-01-27:
You're totally right. This is already accounted in Pinocchio 3x we hope to be able to release soon.

nuft, 2021-01-27:
Glad to hear that it's already being addressed. Looking forward to Pinocchio 3x

closed 2021-01-27 by nuft