stack-of-tasks/pinocchio #1388: Intertial information of links between fixed joints

domire8, 2021-02-04:
Hi, 

I have a  Franka Panda Arm with a F/T sensor and a tool mounted on its end effector flange, as well as the corresponding urdf file. The sensor is attached to the last link by a fixed joint and the tool is attached to the sensor by a fixed joint. Now, to compute the bias of the sensor, I'd need to know the inertial information (center of mass, mass) of the end effector tool.

Unfortunately, it seems that all links between *movable* joints are combined to one element with inertia and the *fixed* joints are only stored as frames (from [here](https://stack-of-tasks.github.io/pinocchio/faq.html) question and [here](https://github.com/stack-of-tasks/pinocchio/issues/802)). Is this correct? Or is there a way of getting the inertial information of links that are connected by fixed joints (in my case the sensor, and end effector tool)?

Or, is this a know issue and you're already working on such a feature? It would be nice to have the complete urdf parsed without loss of information.

jcarpent, 2021-02-04:
> Unfortunately, it seems that all links between _movable_ joints are combined to one element with inertia and the _fixed_ joints are only stored as frames (from [here](https://stack-of-tasks.github.io/pinocchio/faq.html) and [here](https://github.com/stack-of-tasks/pinocchio/issues/802)). Is this correct?

You're correct. We merged the information because it is not needed at the algo level.

> Or is there a way of getting the inertial information of links that are connected by fixed joints (in my case the sensor, and end effector tool)?

It would be indeed possible to store this information, or maybe, a simpler way would be to give to user the access to the original URDF tree info. I think Python bindings of the urdf parser exists and should let you know this information.

Could you explain a bit more why do you need to decouple the inertia of the tools from the end-effector?

eeberhard, 2021-02-04:
Thanks for the prompt reply. I'm working with @domire8 and am happy to explain a bit further. 

We have a tool mounted on a force/torque (FT) sensor, which in turn is mounted to the robot tool flange with an adapter. In the URDF, we define the fixed joint between the flange and the FT sensor (so that we can find the sensor reference frame) and the fixed joint between the FT sensor and the tool tip, in the following tree:
... Robot Joints ... -> Robot Flange -> FT sensor -> Tool tip

To measure contact forces at the tooltip, we need to subtract the gravity force and torque induced by the tool on the force/torque (FT) sensor. For this it would be ideal to use the `centerOfMass()` function and extract the very last subtree (the center of mass of the tool in the frame of the FT sensor). But, as you have noted, the frames following the final (non-fixed) joint are merged into one rigid subtree, so that the last element of `data.com[]` is actually including the robot flange, FT sensor and tool inertia all together.

We were hoping to have the robot and tool definition in a "single source of truth" in the URDF file, and then use the parsed model representation for everything in our control libraries. While it's possible for us to manually parse and supply the tool inertia on the side, it would be much neater if we could find that information from the model for the fixed joint frames in some way.

jcarpent, 2021-02-04:
OK. Thanks for this clear explanation. 
We may imagine extending the current API of Frames, with several types of frames, each one being able to contain some specific but useful information for the end-user.
For instance, BodyFrame containing a rigid body inertia value. SensorFrame containing some sensors information (Camera, etc.).

This feature could be implemented in Pinocchio 3.x. Does it match your calendar?

eeberhard, 2021-02-04:
Thank you, I'm sure such a feature would be welcome to many users. From my experience it's quite common to have a force sensor mounted in series with tooling both in research and industry. 

For a camera or other sensors that are more purely kinematic, the spacial information is already available with e.g. `framesForwardKinematics()`; I'm not sure if there is a need for multiple frame types, as long as each frame has an associated inertial property (that by default is massless). But maybe my reasoning is limited here and having specific FrameTypes would be useful in other ways.

We will gladly follow the development of this feature. In the mean time, we will pass the tool information to our library as a separate parameter so we can proceed with our research.

jmirabel, 2021-02-05:
From an implementation view point, what @jcarpent suggests changes the way Frame are designed. @jcarpent do you have any idea of how this could be implemented ?
With standard inheritance, you would have to stored a vector of pointer to Frame and dynamic_cast them. Is this what you have in mind ?

jcarpent, 2021-02-05:
Variants would be the key, as for Joints. Otherwise, adding an Inertia is reasonably I think. @jmirabel What do you think?

jmirabel, 2021-02-05:
This means another level of complexity with a whole set of new visitors... and it will impact the compilation time.

The frame stuff is not time-critical so I would choose dynamic inheritance. The logic of dynamic casting can be made internal to ModelTpl or another class. This *may* slightly slowdown the update of frame placements because of it adds one more address jump, possibly far in RAM (this can be in the base class no cast involved).

jcarpent, 2021-04-01:
Solved by #1415.

closed 2021-04-01 by jcarpent