stack-of-tasks/pinocchio #1390: problem with XYZQUATToSE3

ahmadgazar, 2021-02-09:
Hi all,
I am testing a simple script to test the transformation from a xyzquaternion to SE3 using pinocchio. The code is as follows:

```
omega = random.rand(3)
R1 = np.array([[1,0,0],[0,1,0],[0,0,1]])     
M1 = pin.SE3(R1, np.zeros(3))     
R2 = R1@pin.skew(omega)
print('R2 = ', R2, '\n')

q2 = Quaternion(R2)
R2_from_quat = pin.XYZQUATToSE3([0.0, 0.0, 0.0,
               q2.w, q2.x, q2.y, q2.z]).rotation
print('R2_from_quat = ', R2_from_quat)
```
From the above code, I was expecting R2 to be equal to R2_from_quat, however, I got the following result:
```
R2 =  [[ 0.         -0.5115753   0.8535087 ]
 [ 0.5115753   0.         -0.74657787]
 [-0.8535087   0.74657787  0.        ]] 

R2_from_quat =  [[ 0.5         0.74657787  0.        ]
 [ 0.74657787 -0.11475703  0.        ]
 [ 0.          0.         -0.61475703]] 
```
Is there something I am missing above? thank you for your help.

jcarpent, 2021-02-09:
`pin.skew` is an antisymmetric matrix, you were much looking for `pin.exp`.

jcarpent, 2021-02-09:
By the way, which class Quaternion are you using. I hope the one provided by Pinocchio.

ahmadgazar, 2021-02-09:
@jcarpent  Oh I see thanks a lot ! Yea, It's the Quaternion class from pinocchio. Also, I noticed that I had to rearrange the quaternion scalar to be in the tail of the quaternion vector, which was a bit confusing to get the correct result.

closed 2021-02-09 by ahmadgazar

jcarpent, 2021-02-09:
in this case, you can extract the coeffs. This follows Eigen conventions and API.