stack-of-tasks/pinocchio #1392: achieving Jacobian in global space for integration with Ceres optimization framework

khanhha, 2021-02-10:
Hello,
I would like to ask about how to achieve Jacobian matrices in global space to be to optimization Pinocchio configuration vector with Ceres.

As shown in this repos, https://github.com/jcarpent/pinocchio-ceres, two main things need to be implemented for integration with Ceres.

The first thing is the CostFunction, which returns Jacobian matrices and residual vectors.
[image]

The second thing is LocalParameterization, as how below. 
[image]

From my experiments, Ceres expects the Jacobian matrix returned by the CostFunction is in global space, but Pinocchio just supports returning Jacobian matrices of the residual vector with respect to tangent space [**nv**, not **nq**]. 

In this example, @jcarpent just copy the residual jacobian matrix of size [residualxnv] to the output jacobian matrix of size [residualxnq], without any transformation.  It seems that it doesn't work in my case.

[image]

So I would like to ask if there is any Jacobian-related function in Pinocchio for calculating Jacobian matrix of size [residualxnq]?

jcarpent, 2021-02-10:
No, you need to work with the tangent spaces and integrator. Ceres is able to handle it and this how you have to work with Ceres. Working with the configuration vector itself is a very bad idea.

jcarpent, 2021-02-10:
You should consider starting from my examples.

khanhha, 2021-02-10:
Thanks for your quick response.

Yes, I actually started from your example before moving into writing a cost function that returns the Jacobian matrix of [2xnq] where 2 is the pixel reprojection error and nq is the size of the configuration vector. Below is my ReprojectionCostFunction. I just add one additional ImageJacobian matrix of size [2x3] to relates changes in image to changes in 3D locations. 

[image]

The thing is that Ceres it seems like my ReprojectionCostFunction returns a local jacobian matrix rather than a global jacobian matrix as Ceres expects. The gradient checker clearly shows that the jacobian returned from ReprojectionCostFunction::evalute is equal to the local numeric jacobian matrix, as shown in the below result from the GradientChecker. The top red line denotes the returned jacobian matrix from the CostFunction, and the bottom red line denotes the local numeric jacobian matrix. The green lines show that the two Jacobian matrices are equal.  This means that my returned Jacobian matrix is equal to the local numeric jacobian matrix; while Ceres expects it to be the global jacobian matrix. 

[image]

Do you have any idea why it's incorrect? I have no clue to continue the debugging now. 

Thank you very much.

jcarpent, 2021-02-10:
My advice would keep using and extending my example, instead of switching to a new one.
In addition, I think your error is not related to Pinocchio itself. It will be hard to provide guidance on your issue.

khanhha, 2021-02-10:
I see. 
Thank you very much for your advice.
I really appreciate it. 
Best
Khanh

closed 2021-02-10 by jcarpent