stack-of-tasks/pinocchio #1393: ReferenceFrame WORLD, LOCAL or LOCAL_WORLD_ALIGNED for achieving Jacobian matrix between Pinocchio tangent space and image space.

khanhha, 2021-02-11:
Hello,

Sorry if I am asking too many questions on this form. Please ignore this question if it's not relevant to Pinocchio.

My problem is finding a Jacobian matrix of size _2xnv_ that relates the changes from Pinocchio tangent space to pixel errors in image space.  For some reason, numeric Jacobian from Ceres's Gradient Checker is different from the one I will explain below. 

As far as I know, the final Jacobian of size  _2xnv_ can be decomposed into a chain of 3 Jacobian matrices: 

`J_Final = Jac_camera_To_Pixel *  Jac_World_To_Cam * Jac_Tangent_To_World`

- _Jac_camera_To_Pixel_ matrix of size _2x3_ relates the changes of a 3D point in camera space to pixel error in image space. A sample code can be found [here](https://github.com/izhengfan/ba_demo_ceres/blob/master/parametersse3.cpp). After many tests, I am 99% sure this code is correct. 
 
- _Jac_World_To_Cam_ of size _6x6_  relates the linear/angular changes of Pinocchio joint in world space to camera space.  Given camera Rt matrix [in OpenCV format] that maps a point from world space to camera space, _Jac_cam_to_world_ can be calculated as below
`
        Eigen::Matrix<double, 3, 3> cameraRot = CamRt.leftCols(3);
        Eigen::Matrix<double, 3, 1> cameraTrans = CamRt.rightCols(1);
        pin::SE3 camSE(cameraRot, cameraTrans);
        Eigen::Matrix<double,6,6> Jac_World_To_Cam;
        pin::Jlog6(camSE, Jac_World_To_Cam);
`

- _Jac_Tangent_To_World_ of size _6xNV_ that relates the changes in tangent space to the world space of a joint. The code is shown below.
`
pin::computeJointJacobians(pinModel,  pinData,  q_vector);
pin::Data::Matrix6x Jac_Tangent_To_World(6,  pinModel.nv);
Jac_Tangent_To_World.fill(0);
pin::getJointJacobian(pinModel, pinData, jointIndex, pin::WORLD, Jac_Tangent_To_World);
`

- The final matrix is calculated as below
`
Eigen::Matrix<double, 2, Eigen::Dynamic, Eigen::RowMajor>  J_Final
J_Final =  Jac_camera * (_Jac_World_To_Cam_*Jac_Tangent_To_World).topRows(3); 
`

I have tried many ways to achieve this matrix but all are wrong, according to [the numeric gradient checker ](https://github.com/jcarpent/pinocchio-ceres/blob/master/examples/inverse-geometry.cpp) from Ceres. The above explanation is the most intuitive answer I can have. 

so my question is: is there anything wrong with my understanding of the way Pinocchio generates Jac_Joint_to_World that causes the incorrect Jacobian matrix?

Thanks for reading until here. 

Sorry if the question is not relevant to Pinocchio. None of my supervisors have knowledge of Pinocchio, so I have no one to count on.

jcarpent, 2021-02-11:
I think Jac_2x3_camera_To_Pixel * Jac_6xNV_Tangent_To_LOCAL_FRAME_OF_THE_CAMERA.topRows(3) should do the job.

khanhha, 2021-02-11:
is it correct that
```c++
Jac_6xNV_Tangent_To_LOCAL_FRAME_OF_THE_CAMERA =  Jac_6x6_World_To_Cam * Jac_6xNV_Tangent_To_World
```

If it's not, can you give me some hint for how to achieve this matrix? _Jac_6xNV_Tangent_To_LOCAL_FRAME_OF_THE_CAMERA_

Thank yo

khanhha, 2021-02-11:
Here is how I calculate _Jac_6xNV_Tangent_To_World_
I assume Rt here is already a relative transformation from world space to camera space, so _pin::Jlog6_ can be used to the corresponding Jacobian matrix. 

```c++
Eigen::Matrix<double, 3, 3> cameraRot = Cam3x4RtMatrix.leftCols(3); 

Eigen::Matrix<double, 3, 1> cameraTrans = Cam3x4RtMatrix.rightCols(1);

pin::SE3 camSE(cameraRot, cameraTrans);

Eigen::Matrix<double,6,6> Jac_6x6_World_To_Cam;

pin::Jlog6(camSE, Jac_6x6_World_To_Cam);
```

jcarpent, 2021-02-11:
J_LOCAL_FRAME_OF_THE_CAMERA = camera_M_joint.action * J_LOCAL_FRAME_OF_THE_JOINT

Sorry @khanhha, but I won't have time to reply further on technical details of that sort, which are out of the scope of Pinocchio. You can read the Handbook of Robotics to learn more on this topic on your own.

khanhha, 2021-02-11:
@jcarpent 
Thank you very much for your help.
That is already tremendously helpful for me.
The issue can be closed here. I wish you all the best.

closed 2021-02-11 by khanhha