stack-of-tasks/pinocchio #1395: time derivative of jacobian

junhyeokahn, 2021-02-15:
Dear pinocchio authors,
I am sorry if this question is duplicated, but I have a question about an API ```getFrameJacobianTimeVariation```.
I have looked through the previously closed issues #901, #1205 but asking this to make sure.

Based on [Featherstone's spatial vector algebra](https://homes.cs.washington.edu/~todorov//courses/amath533/FeatherstoneSlides.pdf) (in p 23), there are two types of acceleration: 1) change of spatial velocity considering a frame fixed in the body (i.e., classical acceleration), and 2) change of spatial velocity considering a frame fixed in the space (i.e., spatial acceleration).
Following this, one dynamics library dart provides two different jacobian dots: 1) [getJacobianClassicDeriv](https://dartsim.github.io/dart/v6.9.5/d3/d19/classdart_1_1dynamics_1_1Skeleton.html#af7f3c1ff7cef2845fb16ec54bd3129ba) and 2) [getJacobianSpatialDeriv](https://dartsim.github.io/dart/v6.9.5/d3/d19/classdart_1_1dynamics_1_1Skeleton.html#a6c26bcc60a7ac9f4fb5844d28a1cecd0). (My apologies for comparing this to other library but I am trying to update my research project codes with Pinocchio.)

Is the function ```getFrameJacobianTimeVariation``` in pinocchio returns the second one?
I assume so based on the comparison of the quality from my side, but want to make sure.
Also, if this is true, is there another API that returns a classical time derivative of jacobian so that I can compute classical acceleration using the formula ```J*qddot + Jdot*qdot```.
I need this term because, in task space control, we are mostly interested in an acceleration of the frame fixed in the body.

jcarpent, 2021-02-16:
I already told you, but you don't need to compute Jdot when dealing with controllers working at the joint acceleration level.
The shortcut is to directly call `forwardKinematics(model,data,q,v,0*v)` and then you will be able to compute your acceleration drift.

I don't see the difference between ` getJacobianClassicDeriv`and `getJacobianSpatialDeriv` in DART, but I do not know well this framework. But, I can directly discuss the difference between classic and spatial acceleration. Indeed, the linear parts of these two are related by:
```
a_c = a_s + w x v
```

junhyeokahn, 2021-02-16:
Thanks for the reply.

> you don't need to compute Jdot when dealing with controllers working at the joint acceleration level.

The reason I want to use ```J_dot``` is to calculate a joint space acceleration given a task space acceleration. Consider a task space controller that you have the desired acceleration in task space (```a_c_des```) and you want to map it to desired acceleration in joint space (```q_ddot_des```) using the formula ```q_ddot_des = pinv(J)(a_c_des - J_dot*q_dot)```.

I have a two-DOF manipulator example to clarify my question.
[image]

The ```J_dot``` quantity I am interested in is the one I derived in the picture (i.e., the jacobian term related to the classical acceleration), but ```getFrameJacobianTimeVariation``` returns the one related to the spatial acceleration.

I know the relationship between ```a_c``` and ```a_s``` but not sure how to relate two jacobians.

jcarpent, 2021-02-16:
I do understand your problem you aim to solve, but again, Jdot*v = data.a[joint_id] after calling `forwardKinematics(model,data,q,v,0*v)`
Please see the code of https://github.com/stack-of-tasks/tsid which implements this kind of controllers.

junhyeokahn, 2021-02-16:
Oh, you're right and I see what you're saying.
So I can basically get the whole ```Jdot*v``` term using ```getFrameClassicalAcceleration``` if I want that this quantity to be related to classical acceleration.
I appreciate your explanation!

closed 2021-02-16 by junhyeokahn

jcarpent, 2021-02-16:
Yes, this trick is simple but very efficient.

whtqh, 2022-10-18:
- [x] using dJ_spatial to calculate dJ_classic, derived from 
```
a_c = a_s + w x v
a_c = J*ddq + dJ_c * dq = J * ddq + dJ_s * dq + cross(J[3:6,:] * dq, J[0:3,:]) * dq
```
dJac6D_classic = pin.frameJacobianTimeVariation(model, data, q, dq, frame_id, pin.WORLD)
dJac6D_classic[0:3,:] += np.dot(pin.skew(np.dot(Jacobian[3:6,:], dq)), Jacobian[0:3, :])