stack-of-tasks/pinocchio #1396: Extracting forward and inverse kinematics quantities of subtrees

ahmadgazar, 2021-02-18:
Hi all,
I am trying to extract forward and inverse kinematics quantities of subtrees of a floating base robot. For example, I would like to get the FK  for each leg, regardless of the contribution of the floating base. Is that available in Pinocchio? Thanks for your help.

jcarpent, 2021-02-18:
@ahmadgazar The first reflex I would recommend it to check in the history of issues, where many people already rase these concerns about subtrees computations. Unfortunately, we are lacking manpower for this interesting and appealing feature, so any contribution from your side is more than welcome. Could you handle it?

But it is easy to find some trick to getting around the issue:
- call classic FK and then remove the subtree root velocity from its children. Same applies to the placement quantities.
- extract a reduced model from the bigger one.

ahmadgazar, 2021-02-18:
I see, thanks the a lot. Indeed, I just checked the history and it seems that it remains an open issue. I would probably be interested, but sorrowfully in the mean time I am quite busy with other projects. I will try to jump in once I allocate more time.

closed 2021-02-18 by jcarpent