stack-of-tasks/pinocchio #1397: what is an effective way to find pseudo inverse of pinocchio::integrateCoeffWiseJacobian

khanhha, 2021-02-18:
Hi,

Recently, I have encountered the problem of integrating Pinocchio with Ceres, which requires ceres::LocalParameterization that computes a Jacobian matrix of size NQxNV for mapping from the tangent space [NQ] to the configuration [or global] space [MV]. 

In addition, the method ceres::CostFunction::Evaluate expects the output Jacobian to be in the configuration space [or global space]; while Pinocchio just provides the joint jacobian matrix of size 6xNV [tangent space].  To handle this problem, in this example, @jcarpent uses a trick to set the first NV columns, as a way to map the tangent-space Jacobian matrix to the global-space Jacobian matrix.   
[image]

For some reason, this trick doesn't work in my case, where I need to find a Jacobian matrix from the image space to the configuration space. A correct solution that I have found is finding the pseudo-inverse of the Jacobian matrix from **integrateCoeffWiseJacobian** and then multiply it with the tangent-space Jacobian matrix

```c++
Jac_2xNV_Image_To_Tangent = J_2x6_cam_angular * Jac_6x6_World_To_Cam * Jac_6xNV_Tangent_To_World
Jac_2xNQ_Image_To_Global = Jac_2xNV_Image_To_Tangent * Jac_NVxNQ_Tangent_To_Global. 
``` 

Here is how Jac_NVxNQ_Tangent_To_Global is calculated
[image]

So my question is that: Pinocchio has the method **integrateCoeffWiseJacobian** for computing the Jacobian matrix of size NQxNV, is there also a more effective method to compute the pseudo-inverse NVxNQ of this matrix?

jcarpent, 2021-02-18:
> For some reason, this trick doesn't work in my case, where I need to find a Jacobian matrix from the image space to the configuration space. A correct solution that I have found is finding the pseudo-inverse of the Jacobian matrix returned by **integrateCoeffWiseJacobian** and then multiply it with the tangent-space Jacobian matrix

I don't understand your internal obligation, and unfortunately, I won't have time to help you on this topic. I think you can take inspiration from what I've done almost 3 years ago. And my only recommendation is that you need to operate on the tangent space all the times. Never compute stuff working on the configuration space. If you need to map from configuration to tangent space, you should use the difference functions and their analytical derivatives that we've implemented with @jmirabel.
For your problem, if I remember well, all the tricks are here: https://github.com/jcarpent/pinocchio-ceres/blob/master/include/pinocchio/ceres/local-parameterization.hpp

PS: I've to admit that optimizing on manifolds is a hard topic, and this is "expected" that you need hours or days to find the magic solution.

khanhha, 2021-02-18:
@jcarpent 
Thank you very much for your answer and all your support through the last few weeks. Actually, I solved the problem by using this method to calculate the Jacobian matrix of size NVxNQ by calculating pseudo-inverse from the Jacobian matrix from integrateCoeffWiseJacobian. 

[image]

I understand your point that it's not a good idea to optimize in the configuration space, but ceres::CostFunction::Evaluate expect the returned Jacobian matrix to be in the global/configuration space, and it seems that your idea in this [example](https://github.com/jcarpent/pinocchio-ceres/blob/master/examples/inverse-geometry.cpp)  seems to work just in your specific case. 

In my case, I need to multiply the 2xNV jacobian matrix with the pseudo-inverse of integrateCoeffWiseJacobian to get GradientChecker return PASSED. 

I hope that this issue will be helpful for anyone working on this problem later. 

Again, I really appreciate your time and your kindness when you reply to my issue at this time.

I wish you all the best.

jcarpent, 2021-02-18:
By the way, implementing the inverse of `integrateCoeffWiseJacobian` should be easy from the current code. You should give it a try I think.

khanhha, 2021-02-18:
Thank you very much.
I'll try it. 
Best

closed 2021-02-18 by khanhha