stack-of-tasks/pinocchio #1400: UserWarning: Error while starting the viewer client.

yl7094, 2021-03-07:
I got the gui opened but the viewer seems have some problem and here is what I get for viwer client

/opt/openrobots/lib/python2.7/site-packages/pinocchio/robot_wrapper.py:238: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
Traceback (most recent call last):

File "/opt/openrobots/lib/python2.7/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 137, in loadViewerModel
    gui = self.viewer.gui
AttributeError: 'GepettoVisualizer' object has no attribute 'viewer'

jcarpent, 2021-03-07:
How did you install Pinocchio and the viewer?
Usually, this error means that you did not open the viewer inside another terminal.

yl7094, 2021-03-07:
Thank you for quick response, I install found I was install from source and when I install by adding robotpkg apt repository, this problem solved

closed 2021-03-07 by yl7094