stack-of-tasks/pinocchio #1406: Path to the URDF files

vpadois, 2021-03-17:
Hello,

I installed pinocchio with conda and started to run the first python examples related to the loading of URDF files description of robots and their visualization. When running the scripts, the retained path seems to be the parent directory of the directory where the python script being run is saved and the URDF files are of course not found there.

How should I proceed (beyond hardcoding the paths) to solve this problem in a clean and permanent way ?

Thanks in advance.
Vincent

vpadois, 2021-03-18:
I found a workaround that will do the trick for now. If it is not the case yet, it would be great if the conda installation could come with the models. If it is the case, please let me know.

jcarpent, 2021-03-18:
Hi Vincent!

It is not clear to me what is not correctly working in your setup, but I do think it is very easy from the current examples to adapt to your local settings. For instance, with the following code:
https://github.com/stack-of-tasks/pinocchio/blob/0856656a329cb53263e8d3e90ee2bb3eec6cf24d/examples/meshcat-viewer.py#L15-L22
You should just provide the mesh_dir location, often the root of your robot_description package.
You can also define the env variable `ROS_PACKAGE_PATH`  to say where your models are located.

Finally, you can install a large variety of robot models by installing the package `example-robot-data`:
```
conda install example-robot-data -c conda-forge
```

vpadois, 2021-03-18:
My problem was not so much on how to specify a path related to my local install but rather to have at hand URDF models that fit the way pinocchio works with URDFs. I was trying to use one of the Panda model I have on my side and some of the scripts would not work with that specific URDF.

Anyway, I just added gepetto models through conda (which I had done "by hand" to fix my problem).

Thanks for the swift feedback. I will close the issue

jcarpent, 2021-03-18:
By the way, if you need more models, I think the best way would be to include them in the `example-robot-data` package, which can be easily installer on any system, without requiring ROS or related framworks.

vpadois, 2021-03-18:
Thanks, I will potentially add my future robot model in this way. I will also suggest people at Auctus to rely on a similar folder organization for robot models.

vpadois, 2021-03-18:
I still have a question. What would be the best way to make sure that locations made in reference to package in URDF files such as in 
`<mesh filename="package://example-robot-data/robots/ur_description/meshes/ur5/visual/base.dae"/>`

are properly resolved ? I would have expected the conda install of example-robot-data to solve this  seamlessly but that's not the case. Does it require a proper environment variable definition as suggested by @jcarpent  ? Or is there any way to do that ?

Thanks in advance and sorry for my naive questions.

jcarpent, 2021-03-18:
@nim65s I think you are the best person to answer to it.

nim65s, 2021-03-18:
example-robot-data comes with a few python helpers, here:
https://github.com/Gepetto/example-robot-data/blob/master/python/example_robot_data/robots_loader.py

This mainly provide a `load('your_model')` function, which handle everything, but if you only need to get the location of the package, we have a `getModelPath(subpath)` helper that look into a few probable locations for your urdf / srdf, and after that the path for the meshes is usually this path + `../..`.

An alternative is to define manually the `ROS_PACKAGE_PATH` environment variable.

vpadois, 2021-03-18:
The python helpers in example-robot-data did the trick and indeed greatly simplify the loading process. Thanks a lot.

closed 2021-03-18 by vpadois