stack-of-tasks/pinocchio #1407: Gepetto-viewer connexion failing

vpadois, 2021-03-18:
My naive discovery of Pinocchio continues. Meshcat visualization works great however I can't get gepetto-viewer to work. Here is the error message I get : 
`(pinocchio) vpadois@ychoux:~/Documents/Softs/Pinocchio_playground$ /home/vpadois/anaconda3/envs/pinocchio/bin/python /home/vpadois/Documents/Softs/Pinocchio_playground/pinocchio_display_model_gepetto.py
/home/vpadois/Documents/Softs/Pinocchio_playground/pinocchio_display_model_gepetto.py:25: UserWarning: Error while importing the viewer client.
Check whether gepetto-viewer is properly installed
  viz.initViewer()
/home/vpadois/Documents/Softs/Pinocchio_playground/pinocchio_display_model_gepetto.py:25: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  viz.initViewer()
Error while loading the viewer model. It seems you should start gepetto-viewer
'GepettoVisualizer' object has no attribute 'viewer'`

gepetto-viewer was installed with conda and I launched gepetto-gui before running the script (the gui opens and does not appear to complain).

Any help will be appreciated.

jmirabel, 2021-03-18:
Can you launch the GUI and try in a Python interpreter:
```python
from gepetto.corbaserver import gui_client
gui = gui_client(window_name="test")
```

vpadois, 2021-03-18:
here is the output:
`(pinocchio) vpadois@ychoux:~/Documents/Enseignement/Cours génériques/QP_ Control
/Notebooks$ ipython
Python 3.8.8 | packaged by conda-forge | (default, Feb 20 2021, 16:22:27) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.21.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from gepetto.corbaserver import gui_client
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-b768ff7dfef9> in <module>
----> 1 from gepetto.corbaserver import gui_client

ModuleNotFoundError: No module named 'gepetto'`

Should I install the corba server version of the gui ?

jcarpent, 2021-03-18:
On which OS are you operating?
For the first time, you should at least init Corba:
```
omniNames -start
```

jcarpent, 2021-03-18:
can you echo $PYHTON_PATH?

vpadois, 2021-03-18:
I am running everything in ubuntu noetic. ROS is installed but not running. I run everything in conda with Python 3.8

PYTHON_PATH is empty.

jmirabel, 2021-03-18:
`omniNames` is now useless. You have to understand why the `gepetto.corbaserver` module is not found. I have no experience with conda.

vpadois, 2021-03-18:
I guess the conda information is not so important. How should I have installed the package providing gepetto.corbaserver ?

jcarpent, 2021-03-18:
conda install gepetto-viewer-corba -c conda-forge

jcarpent, 2021-03-18:
> `omniNames` is now useless. You have to understand why the `gepetto.corbaserver` module is not found. I have no experience with conda.

On mac, I need it to set the first time the correct settings.

vpadois, 2021-03-18:
> conda install gepetto-viewer-corba -c conda-forge

That's what I just dit and that was indeed the problem. I had installed the "non corba" version  of the viewer. It now works (and I did not need to start omniNames explicitly). Thanks again.

closed 2021-03-18 by vpadois