stack-of-tasks/pinocchio #1408: Modeling a quadcopter with Pinocchio

andreadelprete, 2021-03-19:
Hi, 

I was wondering what the best way is to model a drone (e.g. quadcopter) with Pinocchio. The only way I can think of is to use a free-flyer joint, which is fine for representing the state, but doesn't capture the control inputs to the system, which would be four 1d forces applied by the propellers. Clearly I could easily compute manually the resulting wrench (which is the control input expected with a free-flyer joint), but it means I should customize my code for handling this specific system, which is a pity because I have a generic code with which I can usually handle all robots just by changing the urdf file.

Any suggestion?

jcarpent, 2021-03-19:
We should add soon the notion of actuation models which should encompass your case, but also the case of free-floating systems.
Currently, Crocoddyl implements the notion of actuators, but definitely, this should go inside Pinocchio.

cmastalli, 2021-03-19:
Yes, we have implemented a quadrotor actuation model.
You might want to inspect the following code: https://github.com/loco-3d/crocoddyl/blob/master/include/crocoddyl/multibody/actuations/multicopter-base.hpp

jcarpent, 2021-03-22:
I will close this issue. @andreadelprete Feel free to reopen it for further discussions

jcarpent, 2021-03-22:
Related to #440.

closed 2021-03-22 by jcarpent