stack-of-tasks/pinocchio #1411: Estimation of dynamic parameter vector

kunpengyao, 2021-03-22:
Hello, I am new to pinocchio. I would like to write an adaptive controller for my robot model, which requries computing the regressor, Y, and parameter vector, T, such that the torque can be linearized as `tau = Y*T`.
I guess I can access Y by calling `pin.computeJointTorqueRegressor`, right? I wonder if I can also access the complete associated parameter vector T? I'd like to use it as a reference of my estimation. Many thanks.

jcarpent, 2021-03-22:
> I guess I can access Y by calling `pin.computeJointTorqueRegressor`, right?

Yes

>  I wonder if I can also access the complete associated parameter vector T?

There is no direct function for that, but it corresponds to the stack of all the inertia parameters:
```c++
const Eigen::DenseIndex total_size = 10*(model.njoints-1);
Eigen::VectorXd parameters(total_size);
Eigen::DenseIndex slide_id = 0;
for(size_t k = 1; k < model.njoints; ++k)
{
  parameters.segment(slide_id,slice_id+10) = model.inertias[k].toDynamicParameters();
  slide_id += 10;
}
```

kunpengyao, 2021-03-22:
Thanks for the reply! Is it the same method `model.inertias[k].toDynamicParameters()` to call in Python model?
Also, I would like to know if Pinocchio calculates analytical Jacobian associated with different types of orientation parametrization? For example, quaternion.

jcarpent, 2021-03-22:
> Is it the same method `model.inertias[k].toDynamicParameters()` to call in Python model?

Yes

> Also, I would like to know if Pinocchio calculates analytical Jacobian associated with different types of orientation parametrization? For example, quaternion. I noticed there's a type of Jacobian called `Jlog3`, but I could not find its clear explanation in the document...

You should never think of rotations as their parametrized version (e.g. Quaternion, rotation matrices, Angle Axis, etc), but rather as a **Manifold** equipped with its **Tangent Space**. In the case of rotations, this tangent space is nothing more than the space of angular velocities.

Then, `log3` and `exp3` are the difference and integration operation over these spaces. You can then define their Jacobians, knowing that `log3(exp3(w)) = w`.

For further explanations, you should visit the C++ documentation and the literature on the topic.

closed 2021-03-22 by jcarpent