stack-of-tasks/pinocchio #1412: Confusing documentation in computeRNEADerivatives

cmastalli, 2021-03-24:
The Docstring documentation of `computeRNEADerivatives` saids
```python
computeRNEADerivatives( (Model)model, (Data)data, (object)q, (object)v, (object)a) -> tuple :
    Computes the RNEA partial derivatives, store the result in data.dtau_dq, data.dtau_dv and data.dtau_da
    which correspond to the partial derivatives of the torque output with respect to the joint configuration,
    velocity and acceleration vectors.
    
    Parameters:
        model: model of the kinematic tree
        data: data related to the model
        q: the joint configuration vector (size model.nq)
        v: the joint velocity vector (size model.nv)
        a: the joint acceleration vector (size model.nv)
```

However, I don't see data.dtau_da.
I understand that this quantity is the mass matrix, but I wouldn't expect that users know this thing.
Furthermore, it should be clear stated that the information is stored in data.M, right? (actually I am not sure of this despite that it seems obvious to me)

jcarpent, 2021-03-24:
This is related to the output of the function, which is nothing more than a tuple.
I will change it, even if I do think the doc is clear at this level.

closed 2021-03-24 by jcarpent

cmastalli, 2021-03-24:
It says that it stores the result in `data.dtau_da`.
Instead, it should explicitly state that `dtau_da` is stored in `data.M`.

> This is related to the output of the function, which is nothing more than a tuple.

Btw, I didn't understand this comment. Is `computeRNEADerivatives` function returning a tuple with dtau_dq, dtau_dv, and dtau_da? If so, this should also be described in the function. Something like "Note that this functions returns a tuple that contains dtau_dq, ....

jcarpent, 2021-03-24:
This appears in the documentation in Python normally that the output is a tuple.

jcarpent, 2021-03-24:
I changed the doc in #1413 to make the output clearer.