stack-of-tasks/pinocchio #1414: randomConfiguration() generating a zero configuration vector

rshum19, 2021-03-28:
Hi, 

I'm having issues generating a random configuration. I'm using the code below to generate a neutral and random configuration.

```
pinocchio::urdf::buildModel(urdf_filename, model);
data = pinocchio::Data(model);

Eigen::VectorXd q = pinocchio::neutral(model);
Eigen::VectorXd qrand = pinocchio::randomConfiguration(model);
```

However, both `q` and `qrand` are vectors of zeros. I expect `q` to be a vector containing only zeros, but `qrand` I would expect something containing non-zero values. Any idea why this could be happening?

Thank you,

jcarpent, 2021-03-29:
Could you send your URDF file?
For info, randomConfiguration uses the upper and lower joint configuration bounds as bounds for the uniform law.

rshum19, 2021-03-29:
I'm actually using the `simple_humanoid.urdf` in `pinocchio/models/simple_humanoid.urdf` directory. After closer inspection I see that the lower and upper bounds for all joints are equal to 0. I tried with a different urdf model that has non-zero bounds and I do get random configuration. 

That solves my issue. 

Thanks,

closed 2021-03-29 by rshum19