stack-of-tasks/pinocchio #1416: Torque regressor for friction

PuzeLiu, 2021-03-29:
Hi,
I'm trying to do system identification with Pinocchio. However, from the documentation, the computeJointTorqueRegressor only builds the matrix w.r.t the inertia parameters. Is there any interface that can get the matrix for the friction and damping parameters?

Thanks!

jcarpent, 2021-03-30:
Could you provide the equations of what you call friction here?
Otherwise, getting the regressor for the damping is trivial, as it is a diagonal term.

PuzeLiu, 2021-03-30:
Hi, thanks for your quick reply.
What I mean is simply the viscous and coulomb's friction parameter. But I'm not sure if both regressors are diagonal matrices. I think the friction on the later joint should affect the torque of the previous one. Should it be upper triangular matrices?

And another question is that does the friction used in the Pinocchio dynamics? I didn't find any related description on this part

jcarpent, 2021-03-30:
Did you write the equations? I think you will realize it more easily.

jcarpent, 2021-03-30:
> And another question is that does the friction used in the Pinocchio dynamics? I didn't find any related description on this part

We have this feature in Pinocchio 3.x, still not released yet.

PuzeLiu, 2021-03-30:
> Did you write the equations? I think you will realize it more easily.

Yes, you are right.  Thanks for the response!

closed 2021-03-30 by PuzeLiu