stack-of-tasks/pinocchio #1422: Compute Coriolis Matrix Error

EpicDuckPotato, 2021-04-02:
Hi,
I'm trying to compute the coriolis matrix for a floating-base system, and I get the following error:

`/opt/openrobots/include/pinocchio/spatial/symmetric3.hpp:418: pinocchio::Symmetric3Tpl<Scalar, Options> pinocchio::Symmetric3Tpl<Scalar, Options>::rotate(const Eigen::MatrixBase<OtherDerived>&) const [with D = Eigen::Matrix<double, 3, 3>; _Scalar = double; int _Options = 0]: Assertion `isUnitary(R.transpose()*R) && "R is not a Unitary matrix"' failed.`

The configuration consists of position, orientation (quaternion), and two revolute joint angles. I've confirmed that the configuration vector is valid:

0
2.23517e-08
0
-0.707107
0
0
0.707107
0
0

I've been able to compute the mass matrix and gravity just fine, and I've also been able to use rnea. I'm not sure why I'm having issues with the coriolis function.

Thanks,
Anoop

jcarpent, 2021-04-02:
You should normalize your configuration vector:
```
pinocchio::normalize(model,q)
```

closed 2021-04-02 by jcarpent

jcarpent, 2021-04-03:
This is also mean that you are working in devel mode.