stack-of-tasks/pinocchio #1423: buildReducedModel might erase collision_model and intertial information

ntorresalberto, 2021-04-07:
I'm trying to work on a reduced model of the [panda robot](https://github.com/Gepetto/example-robot-data/blob/master/robots/panda_description/urdf/panda.urdf).
I'd like to point out some quirks/problems I'm currently encountering:

1. This is more of a feature request: `RobotWrapper` could provide an API to build a reduced model. I'm not sure I can do it right now but I'd be happy to tackle a PR in the future.
2. I don't know if it actually is a problem, but `buildReducedModel` API doesn't offer a way to pass the `collision_model` from the `RobotWrapper` class. I've noticed `RobotWrapper.visual_model` and `RobotWrapper.collision_model` are both `pinocchio.pinocchio_pywrap.GeometryModel`. Would they provide the same functionality? are they interchangeable?
3. In the example below, I tried to create a `RobotWrapper` from the reduced model and it seems that it has lost the intertial information from my urdf? (or there's some other explanation to why it's now giving my `NaN` values for the gravity)

```python
import example_robot_data
import pinocchio as pin

robot = example_robot_data.load('panda')

randomconfig = pin.randomConfiguration(robot.model)
robot.forwardKinematics(randomconfig)
print('randomconfig:', randomconfig, randomconfig.shape)
g=robot.gravity(randomconfig)
print('model G:', g, g.shape)

# lock 4th joint to reduce dof
jn4_idx = robot.model.getJointId('panda_joint4')
reduced_model, reduced_visual_model = pin.buildReducedModel(
    model=robot.model,
    geom_model=robot.visual_model,
    list_of_joints_to_lock=[jn4_idx],
    reference_configuration=pin.neutral(robot.model))

robot2 = pin.RobotWrapper(model=reduced_model, visual_model=reduced_visual_model)

randomconfig2 = pin.randomConfiguration(robot2.model)
print('randomconfig2:', randomconfig2, randomconfig2.shape)
g=robot2.gravity(randomconfig2)
print('reduced model G:', g, g.shape)
```

And here's the output:
```bash
$ python reduced_panda.py 
randomconfig: [ 2.01874195 -0.3871077   1.67996741 -0.56351698  2.44279775  0.68504683
 -0.97782112  0.03072918  0.01111099] (9,)
model G: [0. 0. 0. 0. 0. 0. 0. 0. 0.] (9,)
randomconfig2: [ 0.32026851 -0.08284433  0.76474584 -0.80239598  1.9198922   2.68362164
  0.0366478   0.02542847] (8,)
reduced model G: [nan nan nan nan nan nan nan nan] (8,)
```

jcarpent, 2021-04-07:
>     3\. In the example below, I tried to create a `RobotWrapper` from the reduced model and it seems that it has lost the intertial information from my urdf? (or there's some other explanation to why it's now giving my `NaN` values for the gravity)

This NAN issue has been fixed on the devel branch. It appears also that the panda3d robot has zero Inertia on all its joints.

jcarpent, 2021-04-07:
>     1. This is more of a feature request: `RobotWrapper` could provide an API to build a reduced model. I'm not sure I can do it right now but I'd be happy to tackle a PR in the future.

This feature is more than welcome. Please proceed, your contribution will be highly appreciated.

jcarpent, 2021-04-07:
> 2\. I don't know if it actually is a problem, but `buildReducedModel` API doesn't offer a way to pass the `collision_model` from the `RobotWrapper` class. I've noticed `RobotWrapper.visual_model` [...]

Similarly to what you did for `visual_model`, you can do the same thing on `collision_model`. Collisions are for collision checkins and visual are just for viewing the robot.

ntorresalberto, 2021-04-07:
> This NAN issue has been fixed on the devel branch. It appears also that the panda3d robot has zero Inertia on all its joints.

Yes, I've noticed.
I'm actually using my own urdf, but the result still stands (it changes from 0 to NaN).
Not trying to pressure or anything but is there a planned release soon or should I setup a development branch?
If so, I would need to compile and install both the cpp and python bindings, right? is there a way to install it inside the conda environment? (I'd like to keep it isolated from my system)

jcarpent, 2021-04-07:
> Not trying to pressure or anything but is there a planned release soon or should I setup a development branch?

A release is planned soon but still need to fix #1420 on Windows.

> If so, I would need to compile and install both the cpp and python bindings, right? is there a way to install it inside the conda environment? (I'd like to keep it isolated from my system)

You can do everything in an isolated conda env, everything works well ;)

ntorresalberto, 2021-04-07:
> Similarly to what you did for `visual_model`, you can do the same thing on `collision_model`. Collisions are for collision checkins and visual are just for viewing the robot.

I don't understand what you mean. 
I can't find a way to pass the `collision_model` to `buildReducedModel`:
```
buildReducedModel(...)
    buildReducedModel( (Model)model, (StdVec_Index)list_of_joints_to_lock, (object)reference_configuration) -> Model :
    buildReducedModel( (Model)model, (GeometryModel)geom_model, (StdVec_Index)list_of_joints_to_lock, (object)reference_configuration) -> tuple :
```

Or do you mean to say that they contain the same information? I understand the API is the same, but maybe they get different data from the urdf and thus end up not being interchangeable (like for instance, paths to the .dae files?)

jcarpent, 2021-04-07:
you can pass it as geom_model, it will then be reduced together with the model.

ntorresalberto, 2021-04-07:
> you can pass it as geom_model, it will then be reduced together with the model.

but in that case I would be losing the visual model information?

jcarpent, 2021-04-07:
No, you just need to do it twice: one for the visual, one for the collision.

jcarpent, 2021-04-08:
Seems to be solved. Feel free to contribute when you want @ntorresalberto.

closed 2021-04-08 by jcarpent