stack-of-tasks/pinocchio #1424: linking error in executable test-cpp-serialization in devel

ntorresalberto, 2021-04-07:
Hello again,
I tried compiling on `devel` branch but I get the following error:

```
[27%] Linking CXX executable test-cpp-serialization
cd /home/user/src/github/pinocchio-fork/build/unittest && /usr/bin/cmake -E cmake_link_script CMakeFiles/test-cpp-serialization.dir/link.txt --verbose=1
/usr/lib64/ccache/c++   -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion   -rdynamic CMakeFiles/test-cpp-serialization.dir/serialization.cpp.o  -o test-cpp-serialization  -Wl,-rpath,/home/user/src/github/pinocchio-fork/build/src:/home/user/.conda/envs/pinocchio/lib ../src/libpinocchio.so.2.5.6 -lboost_unit_test_framework -lboost_filesystem -lboost_system -lboost_serialization /home/user/.conda/envs/pinocchio/lib/liburdfdom_sensor.so /home/user/.conda/envs/pinocchio/lib/liburdfdom_model_state.so /home/user/.conda/envs/pinocchio/lib/liburdfdom_model.so /home/user/.conda/envs/pinocchio/lib/liburdfdom_world.so /home/user/.conda/envs/pinocchio/lib/libconsole_bridge.so.1.0 
/usr/bin/ld: CMakeFiles/test-cpp-serialization.dir/serialization.cpp.o: in function `__static_initialization_and_destruction_0(int, int)':
serialization.cpp:(.text+0x7225): undefined reference to `boost::unit_test::framework::impl::master_test_suite_name_setter::master_test_suite_name_setter(boost::unit_test::basic_cstring<char const>)'
/usr/bin/ld: CMakeFiles/test-cpp-serialization.dir/serialization.cpp.o: in function `boost::unit_test::lazy_ostream::instance()':
serialization.cpp:(.text._ZN5boost9unit_test12lazy_ostream8instanceEv[_ZN5boost9unit_test12lazy_ostream8instanceEv]+0x5): undefined reference to `boost::unit_test::lazy_ostream::inst'
collect2: error: ld returned 1 exit status
make[2]: *** [unittest/CMakeFiles/test-cpp-serialization.dir/build.make:117: unittest/test-cpp-serialization] Error 1
make[2]: Leaving directory '/home/user/src/github/pinocchio-fork/build'
make[1]: *** [CMakeFiles/Makefile2:1882: unittest/CMakeFiles/test-cpp-serialization.dir/all] Error 2
make[1]: Leaving directory '/home/user/src/github/pinocchio-fork/build'
make: *** [Makefile:164: all] Error 2
```

jcarpent, 2021-04-07:
Hard to say anything from your log. You should have look to the git lab ci files and check by yourself.

closed 2021-04-07 by jcarpent

ntorresalberto, 2021-04-07:
I get the same error in `master` branch.
I actually haven't installed boost unit test framework.
Shouldn't cmake fail at the beginning? I see it's not marked as `REQUIRED` [here](https://github.com/stack-of-tasks/pinocchio/blob/17c479e2fe91905122713643f489feb8ddf135a3/unittest/CMakeLists.txt#L41).

In any case, is there a corresponding package in the anaconda repository?
I haven't yet found if there's an equivalent in Fedora.

nim65s, 2021-04-07:
Hi @ntorresalberto,

I didn't test in Fedora 32, but in Fedora 31, pinocchio is building fine on robotpkg: http://robotpkg.openrobots.org/rbulk/robotpkg/math/pinocchio

This is not using conda, but only standard fedora packages, as `boost` and `boost-devel`, but also `urdfdom-devel`and `urdfdom-headers-devel`, and obviously `eigen3-devel`.
For build with the python bindings thanks to eigenpy, you'll need  `boost-python3-devel` and `python3-numpy`.
And for build with collisions thanks to hpp-fcl, you'll also need `assimp` and `assimp-devel`

The procedure is available on https://stack-of-tasks.github.io/pinocchio/download.html, but this wouldn't be compatible with conda.

ntorresalberto, 2021-04-08:
@nim65s Thank you.

It seems there was a conflict between using the system `gcc` and `g++` and the libraries installed inside conda, so installing `gcc_linux-64` and `gxx_linux-64` seems to have solved the compilation issues.

I'm still missing something though as 1 test fails:
```99% tests passed, 1 tests failed out of 116

Total Test time (real) =  20.25 sec

The following tests FAILED:
	 71 - test-py-rpy (Failed)
```

I'm not familiar with `robotpkg` but it seems to manage compilation/installation in a global way? I'm using conda because I'd like to at least try to isolate the environment and possibly be able to easily share it, mixing it with system libraries would defeat the purpose.
I'll update for future reference if I find out what's causing it.

jcarpent, 2021-04-08:
Pinocchio is already in conda by the way. You can look at the recipe to build within the conda-forge.

ntorresalberto, 2021-04-08:
in order for `make install` to use the conda prefix, `cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX` was needed.

> Pinocchio is already in conda by the way. You can look at the recipe to build within the conda-forge.

That could help, do you have a link?
I'm actually trying to build the `devel` branch inside a conda environment.
Maybe I'm troubling myself more than needed. Do you usually compile/test it with system libraries?

jcarpent, 2021-04-08:
A part of Pinocchio's CI is relying on conda, especially the Windows ci. See https://github.com/conda-forge/pinocchio-feedstock/ for the recipe.

jcarpent, 2021-04-08:
> `cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX` was needed.

Indeed, you have several installation of the same package and with conda, you need to specify what is the main workspace.

ntorresalberto, 2021-04-08:
> A part of Pinocchio's CI is relying on conda, especially the Windows ci. See https://github.com/conda-forge/pinocchio-feedstock/ for the recipe.

Thank you, I assume you refer to [these files](https://github.com/conda-forge/pinocchio-feedstock/tree/master/recipe).
Then yeah, it could have helped a bit to have the conda build dependencies.

Should I add these flags to my build then? 
```bash
     -DBUILD_WITH_URDF_SUPPORT=ON
     -DBUILD_WITH_COLLISION_SUPPORT=ON
```

jcarpent, 2021-04-08:
If you need FCL and URDF support, the answer is YES.