stack-of-tasks/pinocchio #1425: cmake error in master branch when compiling from source

ntorresalberto, 2021-04-07:
I've also tried compiling in `master` branch but there seems to be a problem with the optional dependency `eigenpy` (which it seems to find but marks as not found?):

```
 cmake ..
-- Package version (ROS package.xml): 2.5.6
-- Since urdfdom >= 1.0.0, the default C++ standard is C++11. The project is then compiled with C++11 standard.
-- Found Boost: /usr/include (found version "1.69.0") found components: filesystem serialization system 
-- The Python bindings of Pinocchio will be compiled along the main library. If you want to disable this feature, please set the option BUILD_PYTHON_INTERFACE to OFF.
-- PythonLibraryDirs: /home/user/.conda/envs/pinocchio/lib
-- PythonLibVersionString: 3.8.8
-- Python site lib: lib/python3.8/site-packages
-- Found Boost: /usr/include (found version "1.69.0") found components: python38 
-- Boost_PYTHON_LIBRARY: /usr/lib64/libboost_python38.so
-- eigenpy FOUND. eigenpy at /home/user/.conda/envs/pinocchio/lib/libeigenpy.so
CMake Error at cmake/package-config.cmake:84 (find_package):
  Found package configuration file:

    /home/user/.conda/envs/pinocchio/lib/cmake/eigenpy/eigenpyConfig.cmake

  but it set eigenpy_FOUND to FALSE so package "eigenpy" is considered to be
  NOT FOUND.  Reason given by package:

  eigenpy could not be found because dependency Eigen3 could not be found.

Call Stack (most recent call first):
  CMakeLists.txt:136 (ADD_PROJECT_DEPENDENCY)

-- Configuring incomplete, errors occurred!
See also "/home/user/src/github/pinocchio-fork/build/CMakeFiles/CMakeOutput.log".
```

I don't know how relevant it is, but I'm trying to compile in Fedora 32 inside a conda environment.

jcarpent, 2021-04-07:
Your environnement is not clean at all. Please start from a clean distribution.

closed 2021-04-07 by jcarpent

ntorresalberto, 2021-04-07:
Just for future googlers arriving here: The problem was caused by having run `cmake ..` in the `devel` branch and trying to execute it again in `master`.

I don't understand why cmake would retain state but a simple `rm -rf build/` and `mkdir build && cd build && cmake ..` fixed it.

jcarpent, 2021-04-08:
You have a CMakeCache.txt which does the saving.

ntorresalberto, 2021-04-08:
Yes, though in my experience, running cmake multiple times hasn't been a problem and should regenerate the build system just fine.

Surely there might be complex build systems that are the exceptions but it sounds to me that `eigenpy` is being cached wrong, as it was found.