stack-of-tasks/pinocchio #1426: Build reduced model with specific configuration

billusbillus, 2021-04-08:
Hello !

I'm working with the Tiago robot and only using five arm joints. I want thus build the reduced pinocchio model corresponding only to these joints. There is a good example (https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_b-examples_o-reduced-model.html) on how to do that but the configuration to block at is either random or neutral (btw I don't really understand what is a neutral config). I would like to lock my robot at a specific configuration given by joint values (state publisher from gazebo for example). How can I do that ? 

Actually my problem is to understand how I can convert joint values (njoints = 25 on my urdf model) to config vector (nq = 34). And how can the difference between these two numbers be explained ? 

Thanks for you time !

jcarpent, 2021-04-08:
> Actually my problem is to understand how I can convert joint values (njoints = 25 on my urdf model) to config vector (nq = 34). And how can the difference between these two numbers be explained ?

Even if you have 25 joints, some of them may have more than one degree of freedom or may lie on manifolds different from R^1. For instance, a continuous joint can be parametrized by two quantities: `cos(theta),sin(theta)`, meaning that nq=2 and nv=1, because the velocity is only defined by the angular velocity around the joint axis.
We make this distinction to avoid singularity issues.

To go a bit further, for your case, you can define in the buildReducedModel function an argument to provide at which value you want to lock the fixed joints. In addition, the notion of neutral configuration is just the identity element of your configuration space. In the case of continuous revolute joint, it is `1,0`.

billusbillus, 2021-04-08:
Thanks for your quick response ! Ok I understand the difference between nq and nv. 
But I still don't know how to define the reference config. Is there a function buildReducedModel in which I can put directly the joint values (size njoints) ? if not how can I compute the reference config (size nq) ?

billusbillus, 2021-04-08:
Ok actually after thinking about what you said and looking at the urdf, I think I got it. Thanks for the explanation about the parametrization of continuous joints, that's what made me understand !

jcarpent, 2021-04-08:
Seems to be solved!

closed 2021-04-08 by jcarpent