stack-of-tasks/pinocchio #1431: Universe and base_link

billusbillus, 2021-04-12:
Hello, it's me again !

So this time I'm confused about the universe frame and the base_link frame. I can see by displaying every frames of my model that the base_link has an z-offset wrt to the universe frame. 

```
 	- Frame name: base_link paired to (parent joint/ previous frame)(0/22)
with relative placement wrt parent joint:
  R =
1 0 0
0 1 0
0 0 1
  p =      0      0 0.0985
```

But when I'm doing a forward kinematic, and look at the situation of the base_link frame with `data.oMf[baseId].translation()`, I got a null vector `0 0 0`. And every other frame situation (position and orientation with  `data.oMf[frameId].translation()` and  `data.oMf[frameId].rotation()`) is looking good (I'm checking with orocos KDL) but present actually this 0.0985 offset. Why is this offset taken into account for other frames but not for the base_link frame ?

I need actually to get the action matrix between the base_link frame and the operating frame, and this is problematic. Thanks by advance for you help !

jcarpent, 2021-04-12:
Could you send the URDF model of your robot and the current configuration that you are dealing with?

jcarpent, 2021-04-12:
What is your `baseId` value?

jcarpent, 2021-04-12:
No, I did get it. Did you use my Inria address? You can find it on https://jcarpent.github.io

jcarpent, 2021-04-13:
Seems to be solved on @billusbillus side.

closed 2021-04-13 by jcarpent