stack-of-tasks/pinocchio #1432: Computing the COM jacobian in the Local frame.

rshum19, 2021-04-13:
Hi, 

I currently using this to obtain the center of mass jacobian. 
```
Eigen::VectorXd com = centerOfMass(model,data,q);
Eigen::MatrixXd Jcom = jacobianCenterOfMass(model,data,q);
```

However, I'm looking a quick way of obtaining the center of mass jacobian in a `LOCAL` frame instead of the `WORLD` frame. I have been looking from the documentation but could not find a function that this this like `getFrameJacobian(model, data, frame_id, rf, J)`. 

Could you provide some insight in how to obtain this quantity in the `LOCAL` frame?

Thank you,

jcarpent, 2021-04-13:
If you want to express the center of mass in another frame, you just need to translate the quantity:
```
oMi[joint_id].actInv(com)
```
For the center of mass Jacobian, you just need to rotate it:
```
oMi[joint_id].rotation().transpose() * Jcom
```

closed 2021-04-13 by jcarpent

rshum19, 2021-04-13:
As always thank you @jcarpent for your prompt response. The support your provide make this Library so valuable!