stack-of-tasks/pinocchio #1435: Question: what is the meaning of DATA members?

kaixqu, 2021-04-16:
Hi,

Could you please provide more concrete definitions for the following members of Data class?
```
    /// \brief Vector of absolute joint placements (wrt the world).
    PINOCCHIO_ALIGNED_STD_VECTOR(SE3) oMi;

    /// \brief Vector of relative joint placements (wrt the body parent).
    PINOCCHIO_ALIGNED_STD_VECTOR(SE3) liMi;

    /// \brief Vector of absolute operationnel frame placements (wrt the world).
    PINOCCHIO_ALIGNED_STD_VECTOR(SE3) oMf;

    /// \brief Vector of joint placements wrt to algorithm end effector.
    PINOCCHIO_ALIGNED_STD_VECTOR(SE3) iMf;
```
For instance, what is body parent? What is operationnel frame (I think they mean all frames)? What is algorithm end effector?

Moreover, could you please provide a naming scheme for these variables? I guess M is for motion in Plucker notaion. Then what is o, i, li?

Any reference is much appreciated! Thanks!

jcarpent, 2021-04-16:
Did you read the lines above? They correspond to the definition of the quantities, also summarized in the doc: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/structpinocchio_1_1DataTpl.html

You may also read the Featherstone book to understand the whole ideas, at least Chap 2, 3 and 4.
Otherwise, if you think the doc is not precise enough, we would be happy to include your contribution at this stage.

jcarpent, 2021-04-16:
`i` stands for joint id
`li` stands for `lambda(i)` the parent joint id of the joint `i`
`f` means frame
`o` means world
`oMi[k]` stands for the relative transform of the k th joint w.r.t. the world frame.

kaixqu, 2021-04-16:
Many thanks for the quick reply! Now with your comment, I would suggest the following

```
/// \brief Vector of absolute joint placements (wrt the world), size of model.njoints
PINOCCHIO_ALIGNED_STD_VECTOR(SE3) oMi;

/// \brief Vector of relative joint placements (wrt the parent joint), size of model.njoints
///        li stands for lambda(i), the parent joint id of the joint i. -> oMi[i] = oMi[parent] * liMi[i]
PINOCCHIO_ALIGNED_STD_VECTOR(SE3) liMi;

/// \brief Vector of absolute frame placements (wrt the world), size of model.nframes
PINOCCHIO_ALIGNED_STD_VECTOR(SE3) oMf;
```

Still, I still cannot get the meaning for the following part:

```
/// \brief Vector of joint placements wrt to algorithm end effector.
    PINOCCHIO_ALIGNED_STD_VECTOR(SE3) iMf;
```
The size seems to be the same as model.njoints instead of model.nframes (with suffix f)? What actually does it stand for? I have never heard of the terminology, algorithm end effector. Could you please elaborate? Thanks!

jcarpent, 2021-04-16:
> The size seems to be the same as model.njoints instead of model.nframes (with suffix f)? What actually does it stand for? I have never heard of the terminology, algorithm end effector. Could you please elaborate? Thanks!

Yes, you're right that this is not a good terminology. This vector is the relative placement between a joint `i` and a child frame which is assumed to be known. This quantity is mostly used for computing the kinematic Jacobian of a frame when expressed locally.

kaixqu, 2021-04-16:
Actually, I am still confused how that can be done with the index being joint. A joint frame may have many children frame, how does Pinocchio determine which child frame to use?

jcarpent, 2021-04-16:
This is an intermediate quantity for the algorithms, the end-user is not expected to use it ;)
I think this is not worth trying to understand its mean if you don't need it.

kaixqu, 2021-04-16:
Just a suggestion: maybe it's better if you add that to the comment?

Thanks a lot for the quick reply. Have a lovely evening and weekend! :)

closed 2021-04-16 by kaixqu