stack-of-tasks/pinocchio #1441: buildReducedModel doesn't throw an exception with list_of_joints_to_lock.size=0

cmastalli, 2021-04-27:
By inspecting the code, it seems that the following code should throw an exception:
```python
import pinocchio
import example_robot_data

anymal = example_robot_data.load("anymal")
m0 = anymal.model
qref = anymal.q0

m1 = pinocchio.buildReducedModel(m0, [], qref)
```
, furthermore, the output of this code is `False`
```python
m0 == m1
```
and finally, if I use m1 might model predictive controller leads to an erratic behavior.

jcarpent, 2021-04-27:
I don't see any reason for that code to throw.
What do you mean by `inspecting the code`?

jcarpent, 2021-04-27:
> and finally, if I use `m1`, then my predictive controller leads to an erratic behavior.

@cmastalli You are very unclear on what you say. How could you imagine we can guess something from your comments?

cmastalli, 2021-04-27:
>  What do you mean by `inspecting the code`?

This is the code that I referred: https://github.com/stack-of-tasks/pinocchio/blob/a1df23c2f183d84febdc2099e5fbfdbd1fc8018b/src/algorithm/model.hxx#L295

Actually, I am fine with no exception if `m0` is equals to `m1`. What I do not understand if why this leads to a different behavior.
Is there any test that I could do it for helping you to understand this issue?

> @cmastalli You are very unclear on what you say. How could you imagine we can guess something from your comments?

The predictive controller generates a different trotting gait. I could share you a video of both motions (if this is useful).

jcarpent, 2021-04-27:
The code you mention might not throw, because the statement is fulfilled.
You should give me the information that changed between the two models ...

cmastalli, 2021-04-27:
There are a couple of differences between both models:
 1. `m1` doesn't contain `armature`, instead `m0` has it
 2. `m1` and `m0` has different inertias
 
About point 2.
It  is like `m1` ignores that `m0` has a FF joint, but it also modifies other values. With this code:
```python
for i in range(len(m1.inertias)): print m1.inertias[i] == m0.inertias[i]
```
I obtained:
```python
True
False
True
True
False
True
True
False
True
True
False
True
True
False
```
and these are some  values
```python
# m0.inertias[1]
  m = 16.8435
  c = -0.00184194 -0.00122364   0.0506034
  I = 
     0.2185 -0.00145495 -0.00254693
-0.00145495     0.64043 -0.00180478
-0.00254693 -0.00180478    0.624441

# m1.inertias[1]
  m = 33.687
  c = -0.00184194 -0.00122364   0.0506034
  I = 
   0.436999 -0.00291091 -0.00509487
-0.00291091     1.28086 -0.00361056
-0.00509487 -0.00361056     1.24888

# m0.inertias[4]
  m = 0.348375
  c =  0.0588517 -0.0108507 -0.0987049
  I = 
  0.00690838  3.26066e-05   0.00143787
 3.26066e-05   0.00775617 -0.000318279
  0.00143787 -0.000318279   0.00102169

# m1.inertias[4]
  m = 0.489546
  c =  0.0707177 -0.0134891  -0.140861
  I = 
   0.0106694  7.04279e-05   0.00204217
 7.04279e-05    0.0116789 -0.000452645
  0.00204217 -0.000452645   0.00125512
```

jcarpent, 2021-04-28:
Solved by #1442.

closed 2021-04-28 by cmastalli