stack-of-tasks/pinocchio #1446: E: Unable to locate package robotpkg-osg-dae

hjw-1014, 2021-05-17:
I followed the tutorials [here](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_1-directgeom.html). At the **1.0) Tips** section I run `sudo apt install robotpkg-py27-pinocchio robotpkg-ur5-description robotpkg-py27-qt4-gepetto-viewer-corba robotpkg-osg-dae`, and it reports:
```
huangjiawei@huangjiawei-Ubuntu:/opt/openrobots/share/pinocchio$ sudo apt install robotpkg-py27-pinocchio robotpkg-ur5-description robotpkg-py27-qt4-gepetto-viewer-corba robotpkg-osg-dae
[sudo] password for huangjiawei: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package robotpkg-ur5-description
E: Unable to locate package robotpkg-osg-dae

```

I have followed [here](https://stack-of-tasks.github.io/pinocchio/download.html) to install Pinocchio, but I got stuck here, I searched some methods but they doesn't work. Could anyone help me out of here... Thank you so much

nim65s, 2021-05-17:
Hi @hjw-1014,

This tutorial is outdated. We are working on new version. Meanwhile, here is the correct command for this :
```bash
sudo apt install robotpkg-py27-pinocchio robotpkg-py27-example-robot-data robotpkg-py27-qt5-gepetto-viewer-corba
```

The osg-dae plugin is now included in the robotpkg-openscenegraph package. ur-description is now included in example-robot-data.

Also, you probably want to use a python 3 version instead of the 2.7.

hjw-1014, 2021-05-17:
> Hi @hjw-1014,
> 
> This tutorial is outdated. We are working on new version. Meanwhile, here is the correct command for this :
> 
> ```shell
> sudo apt install robotpkg-py27-pinocchio robotpkg-py27-example-robot-data robotpkg-py27-qt5-gepetto-viewer-corba
> ```
> 
> The osg-dae plugin is now included in the robotpkg-openscenegraph package. ur-description is now included in example-robot-data.
> 
> Also, you probably want to use a python 3 version instead of the 2.7.

Hi, thank you so much for your reply. I ran the code you suggested:
`sudo apt install robotpkg-py27-Pinocchio robotpkg-py27-example-robot-data robotpkg-py27-qt5-gepetto-viewer-corba
`

it comes with: 
```Unpacking robotpkg-qt5-osgqt (3.5.7r2) ...
dpkg: error processing archive /tmp/apt-dpkg-install-wn8h4u/20-robotpkg-qt5-osgqt_3.5.7r2_amd64.deb (--unpack):
 trying to overwrite '/opt/openrobots/include/osgQt/Export', which is also in package robotpkg-qt4-osgqt 3.5.7r2
Selecting previously unselected package robotpkg-py27-qt5-gepetto-viewer.
Preparing to unpack .../21-robotpkg-py27-qt5-gepetto-viewer_4.12.0r2_amd64.deb ...
Unpacking robotpkg-py27-qt5-gepetto-viewer (4.12.0r2) ...
Selecting previously unselected package robotpkg-py27-qt5-gepetto-viewer-corba.
Preparing to unpack .../22-robotpkg-py27-qt5-gepetto-viewer-corba_5.6.0_amd64.deb ...
Unpacking robotpkg-py27-qt5-gepetto-viewer-corba (5.6.0) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-wn8h4u/20-robotpkg-qt5-osgqt_3.5.7r2_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)```

And I think I installed py37 and py36 before, so should remove one version of them? 
Thank you so much.

nim65s, 2021-05-17:
Thanks for providing this error, this should not happen. I'll fix this.

To fix the installation on your side, the easiest way is to remove those conflicting packages, and start again:
```bash
sudo apt remove robotpkg-\*
sudo apt install robotpkg-py36-pinocchio robotpkg-py36-example-robot-data robotpkg-py36-qt5-gepetto-viewer-corba
```

hjw-1014, 2021-05-17:
@nim65s Hi, I checked #1066 , it told me I need to use` robotpkg/wip` to install` osg-dae`. So I checked this page(http://robotpkg.openrobots.org/robotpkg-wip.html), but I have two questions:
1. Should I actually use `make update` to compile all the files in `robotpkg/wip` first?
2. I don't fully understand here **Binary packages in robotpkg/wip**, I found `robotpkg_install.conf.5` in `/opt/openrobots/man/man5/robotpkg_install.conf.5`, should I all these:
```
PKG_PATH=http://robotpkg.openrobots.org/wip/packages/bsd/ARCH/pub;
PKG_PATH=http://robotpkg.openrobots.org/packages/bsd/ARCH/pub
```
at here `/opt/openrobots/man/man5/robotpkg_install.conf.5`? Thank you very much for you patience.

nim65s, 2021-05-17:
you don't need robotpkg/wip. All the packages you need are available in the main repository.

hjw-1014, 2021-05-17:
> you don't need robotpkg/wip. All the packages you need are available in the main repository.
> 
> edit: and you don't need osg-dae, please don't try to install it.

Okay, thank you, I just remove robotpkg-\* and install it again.

hjw-1014, 2021-05-17:
Hi, @nim65s . The installation seems to be successful. Then I wanna test it, I follow the old tutorial [here](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_1-directgeom.html), ran **ipython** and wanna test Pinocchio, but it reported error: 

```
huangjiawei@huangjiawei-Ubuntu:/opt/openrobots$ ipython 
Python 2.7.17 (default, Feb 27 2021, 15:10:58) 
Type "copyright", "credits" or "license" for more information.

IPython 5.5.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: from pinocchio.utils import *
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-bc0f5ac1a1bf> in <module>()
----> 1 from pinocchio.utils import *

ImportError: No module named pinocchio.utils

In [2]: 
```

Before I can successfully run this, although with another error when I try load URDF file. Ans it seems like the default python version is 2.7, even I install  `robotpkg-py36-pinocchio`, is this relevant with the default version of my Ubuntu18.04?

Thank you so much for your help and your patience.

nim65s, 2021-05-17:
To import the pinocchio python library, you need:
1. to setup the correct version of the python interpreter:
    ```bash
    sudo apt install ipython3
    ```
2. to configure your python interpreter to find the pinocchio library:
    ```bash
    mkdir -p $(python3 -c 'import site; print(site.USER_SITE)')
    echo "/opt/openrobots/lib/python3.6/site-packages/" >> "$(python3 -c 'import site; print(site.USER_SITE)')/robotpkg.pth"
    ```
 3. try it:
    ```bash
    ipython3
    ```
    
    ```python
    import pinocchio
    ```

hjw-1014, 2021-05-17:
It works! Thank you sooooo much. 

But when I tried to import a urdf, which is in `/opt/openrobots/share/example-robot-data/robots/ur_description/ur5_gripper.urdf`, it reports: `ValueError: The file /opt/openrobots/share/example-robot-data/robots/ur_description/ur5_gripper.urdf does not contain a valid URDF model.`

The whole process like this:

```
In [1]: import pinocchio

In [2]: from pinocchio.robot_wrapper import RobotWrapper

In [3]: URDF = '/opt/openrobots/share/example-robot-data/robots/ur_description/ur5_gripper.urdf'

In [4]: robot = RobotWrapper.BuildFromURDF(URDF)
Error:   File /opt/openrobots/share/example-robot-data/robots/ur_description/ur5_gripper.urdf does not exist
         at line 53 in /build/urdfdom-YMMa9X/urdfdom-1.0.0/urdf_parser/src/model.cpp
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-4-c450fcb9196c> in <module>()
----> 1 robot = RobotWrapper.BuildFromURDF(URDF)

/opt/openrobots/lib/python3.6/site-packages/pinocchio/robot_wrapper.py in BuildFromURDF(filename, package_dirs, root_joint, verbose, meshLoader)
     15     def BuildFromURDF(filename, package_dirs=None, root_joint=None, verbose=False, meshLoader=None):
     16         robot = RobotWrapper()
---> 17         robot.initFromURDF(filename, package_dirs, root_joint, verbose, meshLoader)
     18         return robot
     19 

/opt/openrobots/lib/python3.6/site-packages/pinocchio/robot_wrapper.py in initFromURDF(self, filename, package_dirs, root_joint, verbose, meshLoader)
     19 
     20     def initFromURDF(self,filename, package_dirs=None, root_joint=None, verbose=False, meshLoader=None):
---> 21         model, collision_model, visual_model = buildModelsFromUrdf(filename, package_dirs, root_joint, verbose, meshLoader)
     22         RobotWrapper.__init__(self,model=model,collision_model=collision_model,visual_model=visual_model)
     23 

/opt/openrobots/lib/python3.6/site-packages/pinocchio/shortcuts.py in buildModelsFromUrdf(filename, package_dirs, root_joint, verbose, meshLoader, geometry_types)
     25 
     26     if root_joint is None:
---> 27         model = pin.buildModelFromUrdf(filename)
     28     else:
     29         model = pin.buildModelFromUrdf(filename, root_joint)

ValueError: The file /opt/openrobots/share/example-robot-data/robots/ur_description/ur5_gripper.urdf does not contain a valid URDF model.

In [5]: 
```

And when I tried use tiago urdf which is in` /opt/openrobots/share/example-robot-data/robots/tiago_description/robots/tiago_no_hand.urdf`, it reports:
`ValueError: Mesh package://example-robot-data/robots/tiago_description/meshes/pmb2/base/base_collision.stl could not be found.`

**So the urdf file in example-robot-data, we can not directly use them?** Sorry for my so many questions and thank you very much in advance.

nim65s, 2021-05-17:
We now have shortcuts to do this in example-robot-data, eg.

```python

In [1]: from example_robot_data import load

In [2]: robot = load('ur5_gripper')
```

This will give you a `RobotWrapper` instance from pinocchio with everything automatically set up.

hjw-1014, 2021-05-17:
> We now have shortcuts to do this in example-robot-data, eg.
> 
> ```python
> In [1]: from example_robot_data import load
> 
> In [2]: robot = load('ur5_gripper')
> ```
> 
> This will give you a `RobotWrapper` instance from Pinocchio with everything automatically set up.

Thank you so much, it also works! Okay, so should I also still follow the tutorial [here](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_1-directgeom.html), or you will publish a new tutorial later? Currently, I need to get familiar with Pinocchio so I wanna go through the tutorial. Thank you so much.

nim65s, 2021-05-17:
I think you already got all the errors one should expect from the outdated tutorials, so I guess at this point this is safe to continue :)

In case it's not, uou can find the new version, which is still a work in progress, but with all the new syntax here: https://github.com/nmansard/supaero2021/ ; but it might not be safe either, as this lacks a bit of review, test, bullet-proof and clean ; so expect a few rough edges.

hjw-1014, 2021-05-17:
> I think you already got all the errors one should expect from the outdated tutorials, so I guess at this point this is safe to continue :)
>
> In case it's not, uou can find the new version, which is [...]

Thank you for your information, I will go through it this week.
And another thing relevant with loading robot, how can define the ROS_PACKAGE_PATH? Because I wanna load **Tiago** in **gepetto-gui** , I found here #776 , but I don't find the solution.

nim65s, 2021-05-17:
to load tiago in gepetto-gui, you can also use 

```python
In [1]: from example_robot_data import load

In [2]: robot = load('tiago', display=True)
```

The `ROS_PACKAGE_PATH` environment variable should be mostly useless. But if you want it: `export ROS_PACKAGE_PATH=/opt/openrobots/share`

hjw-1014, 2021-05-17:
Understood, it works, thank you very much. I will go through the tutorials then.

jcarpent, 2021-05-18:
Thanks @nim65s for the quick and precise answers.

closed 2021-05-18 by jcarpent

hjw-1014, 2021-05-20:
Hi, @nim65s sorry, I meet new problems now.
```

In [1]: from example_robot_data import load

In [2]: robot = load('tiago', display=True)
```
This worked on my Ubuntu, but I need to run it in my macOS, it didn't work, I even cannot run 
`from example_robot_data import load
`

So, instead, I tried:
```

from pinocchio.robot_wrapper import RobotWrapper

robot = RobotWrapper.BuildFromURDF(urdf_model_path)
```

it reported:

```
Traceback (most recent call last):
  File "/Users/huangjiawei/Google 云端硬盘/1Master_Thesis/pybullet/move_body.py", line 104, in <module>
    robot = RobotWrapper.BuildFromURDF(urdf_model_path)
  File "/opt/anaconda3/envs/BulletPinocchio/lib/python3.6/site-packages/pinocchio/robot_wrapper.py", line 17, in BuildFromURDF
    robot.initFromURDF(filename, package_dirs, root_joint, verbose, meshLoader)
  File "/opt/anaconda3/envs/BulletPinocchio/lib/python3.6/site-packages/pinocchio/robot_wrapper.py", line 21, in initFromURDF
    model, collision_model, visual_model = buildModelsFromUrdf(filename, package_dirs, root_joint, verbose, meshLoader)
  File "/opt/anaconda3/envs/BulletPinocchio/lib/python3.6/site-packages/pinocchio/shortcuts.py", line 45, in buildModelsFromUrdf
    geom_model = pin.buildGeomFromUrdf(model, filename, geometry_type, package_dirs)
ValueError: Mesh package://example-robot-data/robots/tiago_description/meshes/pmb2/base/base_collision.stl could not be found.
```
`urdf_model_path` is in `models/example-robot-data/robots/tiago_description/robots/tiago.urdf`

I checked here #707, but the solution didn't work for me. 

So in macOS how can I use `RobotWrapper `to build my robot?

hjw-1014, 2021-05-20:
@nim65s Hi, it's weird, yesterday I can run` ipython3 `and` import Pinocchio`, and today I tried again, it reported:

```
huangjiawei@huangjiawei-Ubuntu:/opt/openrobots/share$ ipython3
Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
Type "copyright", "credits" or "license" for more information.

IPython 5.5.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import pinocchio
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-d3c451d02715> in <module>()
----> 1 import pinocchio

/opt/openrobots/lib/python3.6/site-packages/pinocchio/__init__.py in <module>()
      4 
      5 import numpy
----> 6 from .pinocchio_pywrap import *
      7 
      8 from .pinocchio_pywrap import __version__, __raw_version__

ImportError: /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1: undefined symbol: PyClass_Type
```

I searched this error, but I didn't find a solution. Any ideas for me?

nim65s, 2021-05-20:
how did you install the example-robot-data on your macOS ?

hjw-1014, 2021-05-20:
> how did you install the example-robot-data package on your macOS ?

Hi, sorry I figured it out just now, it seems I should use also conda to install example_robot_data, now I can use:

```
In [1]: from example_robot_data import load

In [2]: robot = load('tiago', display=True)
```

to load Tiago again. 

And the other error, I haven't figured it out, I supposed it's relevant with my python version? But I only downloaded the py36 version using the command line.

nim65s, 2021-05-20:
On the other error, I guess you have an issue with  your `$PYTHONPATH`.

Can you check this in a python 3 interpreter ?

```python
import sys
print(sys.path)
```

hjw-1014, 2021-05-20:
@nim65s Hi, the output is:  
```

huangjiawei@huangjiawei-Ubuntu:~$ python3
Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> print(sys.path)
['', '/opt/ros/melodic/lib/python2.7/dist-packages', '/opt/openrobots/lib/python3.6/site-packages', '/home/huangjiawei', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/huangjiawei/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']
>>> 
```

hjw-1014, 2021-05-20:
> This ros melodic thing is in python 2 and has nothing to do here. you are mixing pinocchio in python 2 from ROS and pinocchio in python 3 from robotpkg. This can't work at the moment. and in any case, a python 2 entry showing in a python 3 sys.path will almost always raise this kind of issues.

I accidently added ros path `source /opt/ros/melodic/setup.bash` in `~/.bashrc`, I supposed that's the reason, I deleted it and it works again. Thank you.