stack-of-tasks/pinocchio #1447: code about a cvpr 19 paper

timothy233, 2021-05-17:
Hi, Justin, thanks for the wonderful work on pinocchio. Not sure if it is the correct place to ask my question, but I noticed that the CVPR 19 paper "Estimating 3D Motion and Forces of Person-Object Interactions from Monocular Video" is implemented based on pinocchio. I am wondering if there is a rough timeline for releasing the code. If it still needs time, is there any simple tutorial on inverse dynamics with contacts (for humanoid) to get started? In addition, is it possible to release the urdf file used in the paper for modeling the human?

zongmianli, 2021-05-18:
Hi @timothy233,

Thanks for your interest in our work. Parts of the CVPR project’s code and data are already online but lack documentation and clear instructions. Therefore, we are preparing an official release using Docker for the project. The code release will go with an extended journal version of the paper which should be available around the end of May.
 
> Is it possible to release the urdf file used in the paper for modeling the human?

Our human model is the same used in [HMR](https://akanazawa.github.io/hmr/) but adopted to Pinocchio. The model was not saved in URDF so I would suggest you to wait a few days for the official release.
 
Thanks again for your interest, and I am sorry for the delays!

jcarpent, 2021-05-18:
Thanks @zongmianli for this nice and useful feedback.
@timothy233 Thanks for your interest in our software and research activities.

I will close this issue and we will try to come back to it when the docker image is released.

closed 2021-05-18 by jcarpent