stack-of-tasks/pinocchio #1448: UserWarning: Error while starting the viewer client.

hjw-1014, 2021-05-18:
Hi, I'm new to Pinocchio, and I followed the tutorial [here](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_1-directgeom.html). I already got so much help from @nim65s , and I wanna use `gepetto-gui` to visualize the robot. 

1. What I ran:
`usr_robot = load('ur5_gripper')`. it didn't report error.

`tiago_robot = load('tiago', display=True)`. This report:
```
In [5]: tiago_robot = load('tiago', display=True)
/opt/openrobots/lib/python3.6/site-packages/pinocchio/robot_wrapper.py:272: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
```
I followed here to install the gepetto, what I ran is:
`sudo apt update && sudo apt install robotpkg-py36-qt5-gepetto-viewer-corba`
And it's successful.
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
robotpkg-py36-qt5-gepetto-viewer-corba is already the newest version (5.6.0).
The following packages were automatically installed and are no longer required:
  libomniorb4-2 libomniorb4-dev libomnithread4 libomnithread4-dev libpythonqt-qt5-python2-3 libpythonqt-qt5-python2-dev libpythonqt-qtall-qt5-python2-3
  libpythonqt-qtall-qt5-python2-dev omniidl omniidl-python omniorb-idl omniorb-nameserver python-omniorb python-omniorb-omg

```
**Have I missed something here?** 

2. And when I ran` gepetto-gui`, the gui can be launched but I cannot choose robot from **file**, which is no robot I can choose.
I already set the `export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/opt/openrobots/share`

Thanks in advance.

jcarpent, 2021-05-18:
You should run in another terminal `gepetto-gui`.
Other issues in Pinocchio already mentioned this issue. 
Next time, please look at the history first. It will be quicker and simpler for everybody.

closed 2021-05-18 by jcarpent

hjw-1014, 2021-05-18:
> You should run in another terminal `gepetto-gui`.
> Other issues in Pinocchio already mentioned this issue.
> Next time, please look at the history first. It will be quicker and simpler for everybody.
Thank you so much and sorry for this.

hjw-1014, 2021-05-18:
@jcarpent Hi, sorry to ask this again. I can run the gepetto-gui when I load robot from another terminal. But the second thing I mentioned above:

 when I ran **gepetto-gui**, the gui can be launched but I cannot choose robot from **file** as[ this video](https://homepages.laas.fr/jmirabel/raw/videos/hpp-gui-example.mp4) do, which is no robot I can choose.
I already set the `export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/opt/openrobots/share`

Have I missed something?

nim65s, 2021-05-18:
@hjw-1014 : this is a question on gepetto-viewer., and has nothing to do with pinocchio Please ask it on the right tracker.

hjw-1014, 2021-05-18:
> @hjw-1014 : this is a question on gepetto-viewer., and has nothing to do with pinocchio Please ask it on the right tracker.

Understood, thank you.

jcarpent, 2021-05-18:
@hjw-1014 If you troubles with gepetto-gui, just switch to the Meshcat viewer solution. It is simpler to install and use.

hjw-1014, 2021-05-18:
@jcarpent Thank you very much, I will search Meshchat viewer.