stack-of-tasks/pinocchio #1449: Compiling issue with Boost

ddliu365, 2021-05-19:
Hi,

I need to integrate pinocchio with ROS and met a compiling issue with Boost.

At first a similar error appears like #969 .

I add this one line in the CMakeList.txt at that package.

```
add_definitions("-DBOOST_MPL_LIMIT_LIST_SIZE=30")
```

But it rises another error

```
In file included from /usr/include/boost/variant/variant.hpp:31:0,
                 from /usr/include/boost/variant.hpp:17,
                 from /opt/openrobots/include/pinocchio/multibody/joint/joint-collection.hpp:11,
                 from /opt/openrobots/include/pinocchio/multibody/joint/joint-generic.hpp:8,
                 from /opt/openrobots/include/pinocchio/multibody/model.hpp:17,
                 from /opt/openrobots/include/pinocchio/parsers/urdf.hpp:9,
                 from /home/jack/catkin_ws/src/humanoid/humanoid_estimation_module/include/humanoid_estimation_module/estimation_module.h:9,
                 from /home/jack/catkin_ws/src/humanoid/humanoid_walking_module/include/humanoid_walking_module/walking_module.h:4,
                 from /home/jack/catkin_ws/src/humanoid/humanoid_manager/src/humanoid_manager.cpp:15:
/usr/include/boost/variant/detail/make_variant_list.hpp:61:9: error: wrong number of template arguments (30, should be at least 0)
         >::type type;
         ^
In file included from /usr/include/boost/mpl/aux_/include_preprocessed.hpp:37:0,
                 from /usr/include/boost/mpl/list.hpp:46,
                 from /usr/include/boost/math/policies/policy.hpp:9,
                 from /usr/include/boost/math/policies/error_handling.hpp:19,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/ros.h:38,
                 from /home/jack/catkin_ws/src/humanoid/humanoid_landing_module/include/humanoid_landing_module/landing_module.h:4,
                 from /home/jack/catkin_ws/src/humanoid/humanoid_manager/src/humanoid_manager.cpp:8:
/usr/include/boost/mpl/aux_/preprocessed/gcc/list.hpp:22:8: note: provided for ‘template<class T0, class T1, class T2, class T3, class T4, class T5, class T6, class T7, class T8, class T9, class T10, class T11, class T12, class T13, class T14, class T15, class T16, class T17, class T18, class T19> struct boost::mpl::list’
 struct list;
```

I am using Pinocchio 2.5.6 in Ubuntu 16.04

nim65s, 2021-05-19:
Hi @ddliu365,

Can you check that you always `#include <pinocchio/fwd.hpp>` before anything from boost in all you files ?

ddliu365, 2021-05-19:
Yes, I did
The problem is solved after I adding this to the CMakeLists.txt
```
 target_compile_definitions(${PROJECT_NAME} PRIVATE ${PINOCCHIO_CFLAGS_OTHER})
```

closed 2021-05-19 by ddliu365

roboticszeng, 2024-06-17:
what is PINOCCHIO_CFLAGS_OTHER? @ddliu365