stack-of-tasks/pinocchio #1450: compiling error with ROS

ddliu365, 2021-05-20:
Hi,

I met a compiling error by including pinocchio with ROS melodic 

```
   ^
[100%] Linking CXX shared library /home/jack/catkin_ws/devel/lib/libhumanoid_walking_module.so
[100%] Built target humanoid_walking_module
[100%] Linking CXX executable /home/jack/catkin_ws/devel/lib/humanoid_manager/humanoid_manager
/home/jack/catkin_ws/devel/lib/libhumanoid_estimation_module.so: undefined reference to `pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&)'
collect2: error: ld returned 1 exit status
humanoid/humanoid_manager/CMakeFiles/humanoid_manager.dir/build.make:157: recipe for target '/home/jack/catkin_ws/devel/lib/humanoid_manager/humanoid_manager' failed
make[2]: *** [/home/jack/catkin_ws/devel/lib/humanoid_manager/humanoid_manager] Error 1
CMakeFiles/Makefile2:8128: recipe for target 'humanoid/humanoid_manager/CMakeFiles/humanoid_manager.dir/all' failed
make[1]: *** [humanoid/humanoid_manager/CMakeFiles/humanoid_manager.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
```

Does anyone know how to solve it?

wxmerkt, 2021-05-20:
`target_link_libraries(humanoid_walking_module ${pinocchio_LIBRARIES})` or better `target_link_libraries(humanoid_walking_module pinocchio::pinocchio)`. It's difficult to provide more detailed help without additional detail such as the corresponding CMakeLists excerpt

ddliu365, 2021-05-20:
This solved!

closed 2021-05-20 by ddliu365