stack-of-tasks/pinocchio #1453: cannot install in ubuntu 16.04

thachdo, 2021-05-23:
I followed the installation steps in [here](https://stack-of-tasks.github.io/pinocchio/download.html):
But I cannot any version, errors are as below:

```
apt install robotpkg-py36-pinocchio

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package robotpkg-py36-pinocchio
```
or
```
apt install robotpkg-py27-pinocchio

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 robotpkg-py27-pinocchio : Depends: robotpkg-py27-hpp-fcl (= 1.4.5) or
                                    robotpkg-py27-hpp-fcl (= 1.5.4) or
                                    robotpkg-py27-hpp-fcl (= 1.7.1r1) or
                                    robotpkg-py27-hpp-fcl (= 1.2.1) or
                                    robotpkg-py27-hpp-fcl+doc (= 1.2.0) but it is not going to be installed or
                                    robotpkg-py27-hpp-fcl (= 1.2.2) or
                                    robotpkg-py27-hpp-fcl (= 1.7.1) or
                                    robotpkg-py27-hpp-fcl (= 1.4.2) or
                                    robotpkg-py27-hpp-fcl (= 1.3.0) or
                                    robotpkg-py27-hpp-fcl (= 1.4.4r1) or
                                    robotpkg-py27-hpp-fcl (= 1.2.0) or
                                    robotpkg-py27-hpp-fcl (= 1.6.0) or
                                    robotpkg-py27-hpp-fcl (= 1.4.4) or
                                    robotpkg-py27-hpp-fcl (= 1.7.0)
                           Depends: robotpkg-py27-eigenpy (= 2.2.0) but 2.6.3 is to be installed or
                                    robotpkg-py27-eigenpy (= 2.5.0) but 2.6.3 is to be installed or
                                    robotpkg-py27-eigenpy (= 2.2.0r1) but 2.6.3 is to be installed or
                                    robotpkg-py27-eigenpy (= 2.3.0) but 2.6.3 is to be installed or
                                    robotpkg-py27-eigenpy (= 2.6.1) but 2.6.3 is to be installed or
                                    robotpkg-py27-eigenpy (= 2.4.0) but 2.6.3 is to be installed or
                                    robotpkg-py27-eigenpy (= 2.4.3) but 2.6.3 is to be installed or
                                    robotpkg-py27-eigenpy (= 2.6.2) but 2.6.3 is to be installed
                           Depends: robotpkg-pinocchio (= 2.5.6) but 2.6.0 is to be installed
                           Depends: robotpkg-hpp-fcl+doc (= 1.7.0) but it is not going to be installed or
                                    robotpkg-hpp-fcl+doc (= 1.4.2) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.4.2) or
                                    robotpkg-hpp-fcl (= 1.7.0) or
                                    robotpkg-hpp-fcl+doc (= 1.6.0) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.4.5) or
                                    robotpkg-hpp-fcl+doc (= 1.4.4) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.7.1) or
                                    robotpkg-hpp-fcl+doc (= 1.4.5) but it is not going to be installed or
                                    robotpkg-hpp-fcl+doc (= 1.7.1) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.5.4) or
                                    robotpkg-hpp-fcl+doc (= 1.5.4) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.6.0) or
                                    robotpkg-hpp-fcl (= 1.4.4)
E: Unable to correct problems, you have held broken packages.
```
Even though I tried to `apt update` before installing or install depends like: robotpkg-py27-eigenpy. The problem still cannot be solved. I think the main issue here is
```
Depends: robotpkg-pinocchio (= 2.5.6) but 2.6.0 is to be installed
```
How can I install version 2.5.6 instead of the suggested 2.6.0?

jcarpent, 2021-05-23:
The central question is why you need Pinocchio 2.5.6 and not 2.6.0?
To solve your issue, you should run `apt-get install robotpkg-py35-pinocchio` as Python 3.5 is the official Python 3 version.

jcarpent, 2021-05-23:
By the way, please first uninstall all Pinocchio versions that you have installed so far via apt.

thachdo, 2021-05-23:
`The central question is why you need Pinocchio 2.5.6 and not 2.6.0?`
No, I don't actually need it, just my thinking after taking a look at [here](http://robotpkg.openrobots.org/packages/debian/pub/dists/xenial/robotpkg/binary-amd64/) there is no available version 2.6.0 (the latest one is 2.5.7).

I also obtained the following result with your suggestion:
```
apt-get install robotpkg-py35-pinocchio
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 robotpkg-py35-pinocchio : Depends: robotpkg-py35-hpp-fcl (= 1.2.1) or
                                    robotpkg-py35-hpp-fcl (= 1.4.2) or
                                    robotpkg-py35-hpp-fcl (= 1.6.0) or
                                    robotpkg-py35-hpp-fcl (= 1.2.0) or
                                    robotpkg-py35-hpp-fcl (= 1.7.0) or
                                    robotpkg-py35-hpp-fcl (= 1.2.2) or
                                    robotpkg-py35-hpp-fcl (= 1.3.0) or
                                    robotpkg-py35-hpp-fcl+doc (= 1.2.0) but it is not going to be installed or
                                    robotpkg-py35-hpp-fcl (= 1.4.5) or
                                    robotpkg-py35-hpp-fcl (= 1.7.1) or
                                    robotpkg-py35-hpp-fcl (= 1.5.4) or
                                    robotpkg-py35-hpp-fcl (= 1.4.4) or
                                    robotpkg-py35-hpp-fcl (= 1.4.4r1) or
                                    robotpkg-py35-hpp-fcl (= 1.7.1r1)
                           Depends: robotpkg-py35-eigenpy (= 2.4.0) but 2.6.3 is to be installed or
                                    robotpkg-py35-eigenpy (= 2.2.0r1) but 2.6.3 is to be installed or
                                    robotpkg-py35-eigenpy (= 2.6.2) but 2.6.3 is to be installed or
                                    robotpkg-py35-eigenpy (= 2.6.1) but 2.6.3 is to be installed or
                                    robotpkg-py35-eigenpy (= 2.4.3) but 2.6.3 is to be installed or
                                    robotpkg-py35-eigenpy (= 2.3.0) but 2.6.3 is to be installed or
                                    robotpkg-py35-eigenpy (= 2.2.0) but 2.6.3 is to be installed or
                                    robotpkg-py35-eigenpy (= 2.5.0) but 2.6.3 is to be installed
                           Depends: robotpkg-pinocchio (= 2.5.6) but 2.6.0 is to be installed
                           Depends: robotpkg-hpp-fcl+doc (= 1.7.0) but it is not going to be installed or
                                    robotpkg-hpp-fcl+doc (= 1.4.2) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.4.2) or
                                    robotpkg-hpp-fcl (= 1.7.0) or
                                    robotpkg-hpp-fcl+doc (= 1.6.0) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.4.5) or
                                    robotpkg-hpp-fcl+doc (= 1.4.4) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.7.1) or
                                    robotpkg-hpp-fcl+doc (= 1.4.5) but it is not going to be installed or
                                    robotpkg-hpp-fcl+doc (= 1.7.1) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.5.4) or
                                    robotpkg-hpp-fcl+doc (= 1.5.4) but it is not going to be installed or
                                    robotpkg-hpp-fcl (= 1.6.0) or
                                    robotpkg-hpp-fcl (= 1.4.4)
E: Unable to correct problems, you have held broken packages.
```

thachdo, 2021-05-23:
> By the way, please first uninstall all Pinocchio versions that you have installed so far via apt.

I also tried `sudo apt remove robotpkg-\*` then reinstall but still cannot resolve.

jcarpent, 2021-05-23:
@nim65s Could you help with this issue?

proyan, 2021-05-23:
Hi Van-Thach Do, Did you do sudo apt update first?

thachdo, 2021-05-23:
Hi Rohan, yes I did! I have successfully installed it on ubuntu 18.04 (with ROS melodic) but not 16.04 (no ROS distro). Is there any new update for ubuntu 16.04?

cmastalli, 2021-05-23:
I think Ubuntu Xenial is not longer supported as the distro is deprecated. But @nim65s should confirm this comment as it maintains these binaries.

In general, I think you should move away of 16.04, again this distro is worldwide deprecated.

nim65s, 2021-05-24:
There is a built issue on 16.04 because of a C++98 incompatibility that went unnoticed. While I debug this, you can either:
1. upgrade your 16.04. It is not supported by ubuntu anymore unless you pay for it, so this should clearly be your option
2. keep using robotpkg, but with an older version: 
    ```
    sudo apt update
    sudo apt remove robotpkg-\*
    sudo apt install -qqy robotpkg{,-py35}-pinocchio=2.5.6 robotpkg{,-py35}-hpp-fcl=1.7.0 robotpkg-py35-eigenpy=2.6.2
    ```
3. install the from other binary sources (conda, pip)
4. install from sources

jcarpent, 2021-05-24:
Thanks @nim65s for the prompt answer.

closed 2021-05-24 by jcarpent

thachdo, 2021-05-25:
Thanks @nim65s and @jcarpent! Solved.

thachdo, 2021-05-25:
I tried to build from source and get this issue:

```
pinocchio/build/include/pinocchio/serialization/vector.hpp:108:59: error: invalid initialization of non-const reference of type 'bool&' from an rvalue of type 'std::_Bit_const_iterator::const_reference {aka bool}'
               ar & boost::serialization::make_nvp("item", *hint);
                                                           ^
In file included from /home/ub16_ws/pinocchio/build/include/pinocchio/serialization/fwd.hpp:9:0,
                 from /home/ub16_ws/pinocchio/build/include/pinocchio/serialization/archive.hpp:8,
                 from /home/ub16_ws/pinocchio/build/include/pinocchio/serialization/serializable.hpp:8,
                 from /home/ub16_ws/pinocchio/build/include/pinocchio/multibody/geometry.hpp:12,
                 from /home/ub16_ws/pinocchio/build/include/pinocchio/bindings/python/multibody/geometry-object.hpp:12,
                 from /home/ub16_ws/pinocchio/bindings/python/multibody/expose-geometry.cpp:6:
/usr/include/boost/serialization/nvp.hpp:93:10: note:   initializing argument 2 of 'const boost::serialization::nvp<T> boost::serialization::make_nvp(const char*, T&) [with T = bool]'
 nvp< T > make_nvp(const char * name, T & t){
```

thachdo, 2021-05-25:
I tried to install from your [suggestion](https://github.com/jcarpent/pinocchio/tree/topic/boost) and got new errors:
```
In file included from /home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/model.hpp:17:0,
                 from /home/ub16_ws/patch1/pinocchio/build/include/pinocchio/parsers/urdf.hpp:9,
                 from /home/ub16_ws/patch1/pinocchio/src/parsers/urdf/model.cpp:6:
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:276:93: error: no default argument for 'template<class S, int O> class JointCollectionTpl'
                   const JointDataTpl<Scalar,Options,JointCollectionTpl> & joint_data_generic)
                                                                                             ^
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:276:93: error: no default argument for 'JointDataDerived'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:276:93: error: default template arguments may not be used in function templates without -std=c++11 or -std=gnu++11
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:284:93: error: no default argument for 'template<class S, int O> class JointCollectionTpl'
                   const JointDataTpl<Scalar,Options,JointCollectionTpl> & joint_data_generic)
                                                                                             ^
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:284:93: error: no default argument for 'JointDataDerived'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:284:93: error: default template arguments may not be used in function templates without -std=c++11 or -std=gnu++11
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:291:95: error: no default argument for 'template<class S, int O> class JointCollectionTpl'
                   const JointModelTpl<Scalar,Options,JointCollectionTpl> & joint_model_generic)
                                                                                               ^
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:291:95: error: no default argument for 'JointModelDerived'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:291:95: error: default template arguments may not be used in function templates without -std=c++11 or -std=gnu++11
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:298:95: error: no default argument for 'template<class S, int O> class JointCollectionTpl'
                   const JointModelTpl<Scalar,Options,JointCollectionTpl> & joint_model_generic)
                                                                                               ^
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:298:95: error: no default argument for 'JointModelDerived'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:298:95: error: default template arguments may not be used in function templates without -std=c++11 or -std=gnu++11
In file included from /home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/model.hpp:17:0,
                 from /home/ub16_ws/patch1/pinocchio/build/include/pinocchio/parsers/urdf.hpp:9,
                 from /home/ub16_ws/patch1/pinocchio/src/parsers/urdf/geometry.cpp:5:
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:276:93: error: no default argument for 'template<class S, int O> class JointCollectionTpl'
                   const JointDataTpl<Scalar,Options,JointCollectionTpl> & joint_data_generic)
                                                                                             ^
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:276:93: error: no default argument for 'JointDataDerived'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:276:93: error: default template arguments may not be used in function templates without -std=c++11 or -std=gnu++11
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:284:93: error: no default argument for 'template<class S, int O> class JointCollectionTpl'
                   const JointDataTpl<Scalar,Options,JointCollectionTpl> & joint_data_generic)
                                                                                             ^
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:284:93: error: no default argument for 'JointDataDerived'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:284:93: error: default template arguments may not be used in function templates without -std=c++11 or -std=gnu++11
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/multibody/joint/joint-generic.hpp:291:95: error: no default argument for 'template<class S, int O> class JointCollectionTpl'
                   const JointModelTpl<Scalar,Options,JointCollectionTpl> & joint_model_generic)
                                                                                               ^
[... cut ...]
            for(typename std::vector<T,Allocator>::const_iterator hint = const_value().begin();
                hint != const_value().end(); ++hint)
            {
              ar & boost::serialization::make_nvp("item", const_cast<T&>(*hint));
            }
          }
        }
```
and got
```
/home/ub16_ws/pinocchio/build/include/pinocchio/serialization/vector.hpp:108:50: error: invalid const_cast of an rvalue of type 'std::_Bit_const_iterator::const_reference {aka bool}' to type 'bool&'
               ar & boost::serialization::make_nvp("item", const_cast<T&>(*hint));

```

jcarpent, 2021-05-25:
@nim65s Could you provide some hints here?
@thachdo As 16.04 is now deprecated, I would rather recommend you to switch to Ubuntu 20.04. You won't have any trouble.

jcarpent, 2021-05-25:
@thachdo I've just updated the PR. Could you test it?

thachdo, 2021-05-25:
Thanks for your suggestion, since my current platform is on 16.04. I just want to try my last efforts. If there is no solution then I will consider switching. Just tested your update, the previous errors were removed but:
```
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/bindings/python/multibody/geometry-data.hpp:146:49:   required from here
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:108:50: error: invalid const_cast of an rvalue of type 'std::_Bit_const_iterator::const_reference {aka bool}' to type 'bool&'
               ar & boost::serialization::make_nvp("item", const_cast<T&>(*hint));
                                                  ^
make -f unittest/CMakeFiles/test-cpp-data.dir/build.make unittest/CMakeFiles/test-cpp-data.dir/depend
make[2]: Entering directory '/home/ub16_ws/patch1/pinocchio/build'
```
The casting looks improper.

jcarpent, 2021-05-25:
Did you look at my last commits on #1454?

thachdo, 2021-05-25:
Thanks, still several bugs in the latest commit:

```
/home/ub16_ws/patch1/pinocchio/unittest/serialization.cpp:679:60:   required from here
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:132:56: error: no matching function for call to 'make_nvp(const char [5], std::vector<double>::iterator&)'

/home/ub16_ws/patch1/pinocchio/unittest/serialization.cpp:679:60:   required from here
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:116:57: error: no matching function for call to 'make_nvp(const char [5], std::vector<pinocchio::JointDataTpl<double>, Eigen::aligned_allocator<pinocchio::JointDataTpl<double> > >::const_iterator&)'
               ar & boost::serialization::fixme::make_nvp("item", hint);
                                                         ^

/home/ub16_ws/patch1/pinocchio/unittest/serialization.cpp:679:60:   required from here
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:116:57: error: no matching function for call to 'make_nvp(const char [5], std::vector<pinocchio::MotionTpl<double, 0>, Eigen::aligned_allocator<pinocchio::MotionTpl<double, 0> > >::const_iterator&)'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:28:7: note: candidate: template<class T> const boost::serialization::nvp<T> boost::serialization::fixme::make_nvp(const char*, T*)
       make_nvp(const char* n, T* v) BOOST_NOEXCEP
```

jcarpent, 2021-05-25:
Fixed.

thachdo, 2021-05-25:
Thanks, still have
```
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:116:66: error: lvalue required as unary '&' operand
               ar & boost::serialization::fixme::make_nvp("item", &(*hint));
                                                                  ^
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp: In instantiation of 'void boost::serialization::fixme::nvp<std::vector<_Tp, _Alloc> >::save(Archive&, unsigned int) const [with Archive = boost::archive::xml_oarchive; T = bool; Allocator = std::allocator<bool>]':
/usr/include/boost/serialization/access.hpp:93:9:   required from 'static void boost::serialization::access::member_save(Archive&, T&, unsigned int) [with Archive = boost::archive::xml_oarchive; T = const boost::serialization::fixme::nvp<std::vector<bool> >]'
/usr/include/boost/serialization/split_member.hpp:43:32:   required from 'static void boost::serialization::detail::member_saver<Archive, T>::invoke(Archive&, const T&, unsigned int) [with Archive = boost::archive::xml_oarchive; T = boost::serialization::fixme::nvp<std::vector<bool> >]'
/usr/include/boost/serialization/split_member.hpp:69:18:   required from 'void boost::serialization::split_member(Archive&, T&, unsigned int) [with Archive = boost::archive::xml_oarchive; T = boost::serialization::fixme::nvp<std::vector<bool> >]'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:136:9:   required from 'void boost::serialization::fixme::nvp<std::vector<_Tp, _Alloc> >::serialize(Archive&, unsigned int) [with Archive = boost::archive::xml_oarchive; T = bool; Allocator = std::allocator<bool>]'
/usr/include/boost/serialization/access.hpp:118:9:   required from 'static void boost::serialization::access::serialize(Archive&, T&, unsigned int) [with Archive = boost::archive::xml_oarchive; T = boost::serialization::fixme::nvp<std::vector<bool> >]'
/usr/include/boost/serialization/serialization.hpp:69:22:   [ skipping 21 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/serializable.hpp:72:44:   required from 'void pinocchio::serialization::Serializable<Derived>::saveToXML(const string&, const string&) const [with Derived = pinocchio::GeometryData; std::__cxx11::string = std::__cxx11::basic_string<char>]'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/bindings/python/serialization/serializable.hpp:41:26:   required from 'void pinocchio::python::SerializableVisitor<Derived>::visit(PyClass&) const [with PyClass = boost::python::class_<pinocchio::GeometryData>; Derived = pinocchio::GeometryData]'
/usr/include/boost/python/def_visitor.hpp:31:9:   required from 'static void boost::python::def_visitor_access::visit(const V&, classT&) [with V = boost::python::def_visitor<pinocchio::python::SerializableVisitor<pinocchio::GeometryData> >; classT = boost::python::class_<pinocchio::GeometryData>]'
/usr/include/boost/python/def_visitor.hpp:67:34:   required from 'void boost::python::def_visitor<DerivedVisitor>::visit(classT&) const [with classT = boost::python::class_<pinocchio::GeometryData>; DerivedVisitor = pinocchio::python::SerializableVisitor<pinocchio::GeometryData>]'
/usr/include/boost/python/class.hpp:226:9:   required from 'boost::python::class_<T, X1, X2, X3>::self& boost::python::class_<T, X1, X2, X3>::def(const boost::python::def_visitor<Derived>&) [with Derived = pinocchio::python::SerializableVisitor<pinocchio::GeometryData>; W = pinocchio::GeometryData; X1 = boost::python::detail::not_specified; X2 = boost::python::detail::not_specified; X3 = boost::python::detail::not_specified; boost::python::class_<T, X1, X2, X3>::self = boost::python::class_<pinocchio::GeometryData>]'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/bindings/python/multibody/geometry-data.hpp:146:49:   required from here
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:116:66: error: lvalue required as unary '&' operand
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp: In instantiation of 'void boost::serialization::fixme::nvp<std::vector<_Tp, _Alloc> >::load(Archive&, unsigned int) [with Archive = boost::archive::text_iarchive; T = bool; Allocator = std::allocator<bool>]':
/usr/include/boost/serialization/access.hpp:101:9:   required from 'static void boost::serialization::access::member_load(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = boost::serialization::fixme::nvp<std::vector<bool> >]'
/usr/include/boost/serialization/split_member.hpp:54:32:   required from 'static void boost::serialization::detail::member_loader<Archive, T>::invoke(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = boost::serialization::fixme::nvp<std::vector<bool> >]'
/usr/include/boost/serialization/split_member.hpp:69:18:   required from 'void boost::serialization::split_member(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = boost::serialization::fixme::nvp<std::vector<bool> >]'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:136:9:   required from 'void boost::serialization::fixme::nvp<std::vector<_Tp, _Alloc> >::serialize(Archive&, unsigned int) [with Archive = boost::archive::text_iarchive; T = bool; Allocator = std::allocator<bool>]'
[... cut ...] inocchio::GeometryData>]'
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/bindings/python/multibody/geometry-data.hpp:146:49:   required from here
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:116:66: error: lvalue required as unary '&' operand
               ar & boost::serialization::fixme::make_nvp("item", &(*hint));
                                                                  ^
make[2]: Leaving directory '/home/ub16_ws/patch1/pinocchio/build'
```

jcarpent, 2021-05-25:
I've provided a fix.

thachdo, 2021-05-25:
Thanks, please see
```
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:116:66: error: lvalue required as unary '&' operand
               ar & boost::serialization::fixme::make_nvp("item", &(*hint));
                                                                  ^
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:132:65: error: taking address of temporary [-fpermissive]
             ar >> boost::serialization::fixme::make_nvp("item", &(*hint));
                                                                 ^
```

jcarpent, 2021-05-25:
Sorry, I missed a commit. Could you retry?

thachdo, 2021-05-25:
Thanks, can you recheck again?
```
/home/ub16_ws/patch1/pinocchio/bindings/python/multibody/expose-model.cpp:15:34:   required from here
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/serialization/vector.hpp:120:70: error: invalid const_cast from type 'boost::serialization::fixme::nvp<std::vector<pinocchio::FrameTpl<double>, Eigen::aligned_allocator<pinocchio::FrameTpl<double> > > >::const_iterator* {aka __gnu_cxx::__normal_iterator<const pinocchio::FrameTpl<double>*, std::vector<pinocchio::FrameTpl<double>, Eigen::aligned_allocator_indirection<pinocchio::FrameTpl<double> > > >*}' to type 'boost::serialization::fixme::nvp<std::vector<pinocchio::FrameTpl<double>, Eigen::aligned_allocator<pinocchio::FrameTpl<double> > > >::iterator* {aka __gnu_cxx::__normal_iterator<pinocchio::FrameTpl<double>*, std::vector<pinocchio::FrameTpl<double>, Eigen::aligned_allocator_indirection<pinocchio::FrameTpl<double> > > >*}'
```

jcarpent, 2021-05-25:
Could you try the new fix?

jcarpent, 2021-05-25:
Just to let you know I'm working blindly as I don't have any Ubuntu 16.04 environment.

thachdo, 2021-05-25:
Don't worry, I appreciate your help!
The problem with `vector.hpp` has gone, now we have
```
/home/ub16_ws/patch1/pinocchio/bindings/python/algorithm/expose-model.cpp:113:16:   required from here
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/algorithm/model.hxx:476:74: error: in C++98 'temp_input_geoms' must be initialized by constructor, not by '{...}'
     const std::vector<GeometryModel> temp_input_geoms { input_geom_model };
                                                                          ^
/home/ub16_ws/patch1/pinocchio/build/include/pinocchio/algorithm/model.hxx:476:74: error: no matching function for call to 'std::vector<pinocchio::GeometryModel>::vector(<brace-enclosed initializer list>)'
```

nim65s, 2021-05-25:
For this one, an easy workarong is to compile in C++11:
`cmake -DCMAKE_CXX_STANDARD=11 ..`

jcarpent, 2021-05-25:
This is not really a fix ;). I would prefer to fix the C++98 stuff, as we are still sticking to this until Pinocchio 3.x

jcarpent, 2021-05-25:
@thachdo I've provided a fix.

thachdo, 2021-05-25:
Thanks a lot @jcarpent and @nim65s, both solutions worked! But I prefer the one at the latest commit. Finally, I can successfully build the pinocchio with python36 on ubuntu 16.04. Sorry for the late reply because I need to remove the default libboost version and install a new one along with python3.6.

nim65s, 2021-05-25:
Python 3.6 on 16.04 ? This doesn't look right.

16.04 provides Python 3.5, and the `libboost-python1.58-dev` package correctly contains  `/usr/lib/x86_64-linux-gnu/libboost_python-py35.so`.

If you installed another python version on top of your system, yes, you'll also need to setup a correct boost environment. But in that case, you could get a more recent Boost version, and the issues you got should have been fixed upstream.