stack-of-tasks/pinocchio #1455: What is the definition of joint Jacobian and frame Jacobian?

kaixqu, 2021-05-29:
Hi,

I am recently trying to use the Jacobian function from Pinocchio. But I must say, the documentation is quite misleading to me. I tried to look into the issues, but still, things are still not that clear.

I think when saying Jacobian (or geometric Jacobian), we mean the relations between the velocity of the end-effector and the velocity of the joints that supports this end-effector. 

For instance, if we have an 8-DoF manipulator, then the Jacobian should be of size 6 * 8, where column j corresponds to the velocity of the end-effector when joint j has unit velocity. In other words, for a manipulator with the following kinematic tree structure (0 -> 1 -> 2 -> ... -> 7 -> 8 -> End-effector), then there is only one Jacobian that express how the joint affects the leaf (ee). 

My confusion is then why do we have `computeJointJacobian` whose functionality is `Computes the Jacobian of a specific joint frame expressed in the local frame of the joint and store the result in the input argument J`.

And then, what is the purpose of `getJointJacobian` and `getFrameJacobian`? How are they related? The former one `Computes the Jacobian of a specific joint frame expressed either in the world (rf = WORLD) frame or in the local frame (rf = LOCAL) of the joint`, and the latter one `Computes the Jacobian of a specific Frame expressed in the desired reference_frame given as argument`.

From what I learned, I only know the Jacobian that I described, and by the name of the Pinocchio function, I guess I should then call `getFrameJacobian`? 

In this issue: https://github.com/stack-of-tasks/pinocchio/issues/418#issuecomment-347791826, there is a comment "There is no sense in having getFrameJacobian() in world frame, as, in the world frame, joint jacobian and frame jacobian are equal." Why is that? What is the definition of Jacobian?

Thank you in advance for your clarification!

proyan, 2021-05-31:
Hi Kaixian, 
It looks like the main problem with understanding the documentation that you have, is 
1) between API intended with `compute` and `get` functions
2) Use of the world `frame` to denote the "frame of reference" and also the [class](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/structpinocchio_1_1FrameTpl.html) which denotes some required placements/inertias on the kinematic tree.

Let's look at point (2) first.
The Frame Class in pinocchio denotes a placement (with or without associated inertia) that is `FIXED` w.r.t. a parent joint. So the kinematic tree is composed of a chain of joints, and the Frame objects are `FIXED` on top of individual joints. 

Now let's look at point (1):
If you are given a configuration `q`, then in order to `compute` the jacobian, you would have to go through all the parents, get their placements, and use them to compose the `J` matrix. To do this, you go through the kinematic tree of the robot.

Now, imagine that you have already done that for a joint `j`. If the joint `j` is the parent of a Frame object `f`, and I ask you to `get` me the jacobian at the placement of the frame `f`. Since you already went through the kinematic tree once when `computing` J, it is redundant for you to do that. The jacobian at the frame `f` can be found by only looking at `J` and the relationship between Frame `f` and Joint `j`, i.e. jMf (the relative placement of Frame `f`  w.r.t. Joint `j`).

This is the difference between `compute` and `get`. You need `compute` when you need to use the configuration variable `q`, and you use `get` to obtain values that were computed by `compute`

There are the following functions that are defined for dealing with jacobians:
1) [computeJointJacobians](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#aa94eb0eb9e51486f618e49617a724d9f) : computes ALL jacobians
2) [computeJointJacobian](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#af1948cb50a9211fd338feb2a3d42d661): computes only one joint jacobian
3) [computeFrameJacobian](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen- html/namespacepinocchio.html#a7b1189aea93a252f4c06d9a7e3c9f5e7): computes only one frame jacobian
4) [getJointJacobian](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#abadfcd21d13fa17865a189e4b47acca9): If `computeJointJacobians` is already called, it gives the joint jacobian given by joint id j
5) [getFrameJacobian](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#a5aa19d265b05aaa782bb24ae4d8894f0): If `computeJointJacobians` is already called, it gives the frame jacobian given by frame id f.

The `compute` functions require `q`, `v` and so on to compute the values. The `get` functions normally only require the internal model information (such as "which joint id" or "which frame id"). 

I hope that is clear enough

kaixqu, 2021-05-31:
Hi Rohan, 

Thanks for the quick and detailed reply! So I assume what I described above should be done by first calling `computeJointJacobians`, and then `getFrameJacobian` with the `frameId` of that specific End-effector frame. Please correct me if I am wrong. 

`jointJacobian` refers to the Jacobian of each joint frame, whereas the `frameJacobian` is the Jacobian of all the frames.

I agree that the Jacobian of each joint frame is enough for calculating the Jacobian of its children frame, because the velocity of that joint frame is enough for calculating the velocity of every frame fixed w.r.t. that link (assuming that link is a rigid body). I guess this calculation is done in `getFrameJacobian`. 

However, why this [comment](https://github.com/stack-of-tasks/pinocchio/issues/418#issuecomment-347791826) says "There is no sense in having getFrameJacobian() in world frame, as, in the world frame, joint jacobian and frame jacobian are equal." Why is that? How could the Jacobian of joint frame be the same as that of its children frame? Is there something I am missing?

Thanks! :)

proyan, 2021-05-31:
1) You need to call computeJointJacobians before you call a `get` function.
2) You would have to define the end-effector frame if it is not yet defined. If your "end-effector frame" co incides with the joint (e.g., the wrist of a robot), then you could simply use the `getJointJacobian`, otherwise you would need `getFrameJacobian`
3) **Only for the case of WORLD [reference frame](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/group__pinocchio__multibody.html#gaa56acd11081bc4603c21de42b9c46c81)** , there is the following relation:
`getFrameJacobian(model, data, frame_id, WORLD) == getJointJacobian(model, data, joint_id, WORLD)`, if `joint_id` corresponds to the parent joint of frame `frame_id`. 

As I told you above, the frame is `FIXED` w.r.t the parent joint. If you do the math (I won't do it here), it is easy to see that when looking through the WORLD frame of reference, `jMf` does not play any role in the computation of the jacobian. However, the following relation isn't true when looking through `LOCAL` and `LOCAL_WORLD_ALIGNED` frames.

> How could the Jacobian of joint frame be the same as that of its children frame? Is there something I am missing?

The "jacobian of the joint frame" is not what we are talking about here. What we are talking about is "the jacobian of a frame which is fixed w.r.t a joint" vs "jacobian of the joint". And that too in a specific ReferenceFrame WORLD. See my point (2) above about the difference between a "frame of reference", i.e. [reference frame](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/group__pinocchio__multibody.html#gaa56acd11081bc4603c21de42b9c46c81) and the Frame Class.

kaixqu, 2021-05-31:
First, I have to say I am really confused about your statement on

> `jMf` does not play any role in the computation of the jacobian

Without copying equations from lecture note or somewhere else, can we just look at a 2D inverted pendulum (with only one joint hinged at the origin). The Jacobian of that joint frame should be something like `[0; 0; 1],` as the origin of the joint is not moving. However, the tip of that inverted pendulum definitely have something else for translational part, instead of just zeros. And this is definitely related to `jMf`, or just the length of that pendulum.

Last but not the least, you said that `jointJacobian` is not the "jacobian of the joint frame". Could you please tell me the definition of joint Jacobian (jacobian of the joint) and frame Jacobian? I could not find any definition for it online. From my point of view, jacobian provides the relation between joint velocities and end-effector velocities.

Thanks!

jcarpent, 2021-05-31:
> However, why this [comment](https://github.com/stack-of-tasks/pinocchio/issues/418#issuecomment-347791826) says "There is no sense in having getFrameJacobian() in world frame, as, in the world [...]

I fact, when computing the Jacobians in the WORLD frame, meaning the velocity field associated to the point coincident with the world center, but moving with the body, both Frame and Joint Jacobians expressed in the WORLD coordinate system will refer to the same quantity.

proyan, 2021-05-31:
@kaixqu 

I feel that we are going in circles a bit. I suggest you give it some time to sink in, try a bit in python (it is easy), and then come back. Here is a script for you to play:
```python
import example_robot_data
import pinocchio
robot = example_robot_data.loadTalos()
pinocchio.computeJointJacobians(robot.model, robot.data, pinocchio.randomConfiguration(robot.model)) # compute jacobians for a random configuration
Jf = pinocchio.getFrameJacobian(robot.model, robot.data, 100, pinocchio.WORLD)
Jj = pinocchio.getJointJacobian(robot.model, robot.data, robot.model.frames[100].parent, pinocchio.WORLD)
(Jj == Jf).all()
```
I'll close this issue

closed 2021-05-31 by proyan

XuanLin, 2023-09-24:
> > However, why this [comment](https://github.com/stack-of-tasks/pinocchio/issues/418#issuecomment-347791826) says "There is no sense in having getFrameJacobian() in world frame, as, in the world frame, joint jacobian and frame jacobian are equal." Why is that? How could the Jacobian of joint frame be the same as that of its children frame? Is there something I am missing?
> 
> I fact, when computing the Jacobians in the WORLD frame, meaning the velocity field associated to the point coincident with the world center, but moving with the body, both Frame and Joint Jacobians expressed in the WORLD coordinate system will refer to the same quantity, as mentioned by @proyan.
> 
> This mixing does not appear when one expresses the quantity in their LOCAL or LOCAL_WORLD_ALIGNED coordinate systems.

Hi,

My apologies for posting a comment in this closed issue. I want to check if my understanding is correct. In the figure below, I draw a 2-dof manipulator with theta_1 and theta_2, and three frames: joint JxJy, user-defined FxFy, and world WxWy. For joint Jacobians, we use V_J (V_J=J*dq). For frame Jacobians, we use V_F. My understanding for WORLD Jacobian is, we imagine expanding the link to cover the origin of the world, then use the velocity V_W expressed in the world frame WxWy. On the other hand, for LOCAL, we use V_J or V_F (depending on joint or frame Jacobian), expressed in local frame (JxJy or FxFy). For LOCAL_ALIGNED, we use V_J or V_F, but still expressed in WORLD frame. Is that correct?

[image]