stack-of-tasks/pinocchio #1456: Operation space control

PrinceYueLu-Wang, 2021-06-01:
Hi, thank you for your work, currently I'm doing my thesis using Pinocchio tool. I start from a simple task, operation space control(just move the end-effector from one point to a goal point). I also reference #1205 and #802. So how does Pinocchio change the joints configuration `q`? 
From my understanding, it's pin.forwardkinematics and updateFrameKinematics.
I run the following code, but I found `q `and `dq `never change, can you give me a hint why?

Thans in advance.
Jiawei

```

#
robot = load_tiago(urdf_filename=urdf_filename) # function I wrote the load my own urdf
print('robot.model: ', robot.model)

#
q = pin.neutral(robot.model)
print('q: ', q)
dq = np.zeros(robot.nv)
ddq = np.zeros(robot.nv)
tau = np.zeros(robot.nv)

# desired end-effector
qua = pin.Quaternion(.4, .02, -.5, .7)
print('qua', qua)
qua.normalize()
print('qua', qua)
x_des = pin.SE3(qua, np.array([.2, -.4, .7]).T)
print(x_des)

# Gains
Kp = 1.0
Kd = 0.01

#Repaet the following in the loop

for k in range(10000):
    # compute task space error
    EE_idx = robot.model.getFrameId('arm_7_link')
    print(EE_idx)
    dMi = x_des.actInv(robot.data.oMf[EE_idx])
    dx = pin.log6(dMi).vector
    xd = pin.getFrameVelocity(robot.model, robot.data, EE_idx, pin.ReferenceFrame.WORLD)
    dxd = xd.vector
    xdd_des = Kp * dx - Kd * dxd

    # update jacobians
    pin.computeJointJacobians(robot.model, robot.data, q)
    pin.framesForwardKinematics(robot.model, robot.data, q)
    J = pin.getFrameJacobian(robot.model, robot.data, EE_idx, pin.ReferenceFrame.WORLD)
    pin.computeJointJacobiansTimeVariation(robot.model, robot.data, q, dq)
    dJ = pin.getFrameJacobianTimeVariation(robot.model, robot.data, EE_idx, pin.ReferenceFrame.WORLD)

    # compute feedforward torque/acceleration
    J_pinv = np.linalg.pinv(J)
    qdd_des = np.dot(np.linalg.pinv(J), (xdd_des - np.dot(dJ, dq)))
    pin.computeAllTerms(robot.model, robot.data, q, dq)

    # advance dynamics
    dt = 0.01
    dq += ddq * dt
    q = pin.integrate(robot.model, q, dq * dt)
```

jcarpent, 2021-06-01:
You forgot to apply FK kinematics before computing your reference errors.

jcarpent, 2021-06-01:
By the way, your issue is not really related to Pinocchio. Do you agree?
I think your thesis advisor is much more qualified than us to provide you with some guidance on this topic.

jcarpent, 2021-06-01:
You have also examples in the examples forlder. Please visit them first.

closed 2021-06-01 by jcarpent

PrinceYueLu-Wang, 2021-06-01:
@jcarpent Thank you, I'm just confused about why the q and dq don't update, which I thought I may misunderstand something for the mechanism of Pinocchio. 
Sorry for this and thank you very much  for your explain, I will check the examples to get more understanding about the whole process.

jcarpent, 2021-06-01:
Thanks for this feedback.
This is the fix for your issue is here:

> You forgot to apply FK kinematics before computing your reference errors + you SHOULD use qdd_des in your update formula.

PrinceYueLu-Wang, 2021-06-01:
> Thanks for this feedback.
> This is the fix for your issue is here:
> 
> > You forgot to apply FK kinematics before computing your reference errors + you SHOULD use qdd_des in your update formula.

Received it, thank you very much for your patience and your help, I will look into more examples.

RolandZhu, 2022-04-25:
Hi, just curious whether your code for jacobian and dot_jacobian gives the correct task space behavior? I'm using a similar controller for an RL project.