stack-of-tasks/pinocchio #1457: Model created using `buildReducedModel` produces unexpected forward/inverse dynamics results

nielsvd, 2021-06-07:
In the current `master` branch of Pinocchio, as well as in the latest `ros` package of Pinocchio (2.6.0) the forward and inverse dynamics are not working as expected by us. The accelerations and torques are slightly off from what Pinocchio produced before and what a static baseline based on RBDL returns.

Up to version 2.5.6 of Pinocchio the forward/inverse dynamics were as we expected. We have traced back that the change in behavior was introduced in this commit: https://github.com/stack-of-tasks/pinocchio/commit/483dcca8eb6b8481fa3115ad743d8a3ad74951c0. Before this commit, the behavior is as it was before, after it as we observe in the current state of the `master` branch.

We also observed that the problem does not exist (anymore) in the current `devel` branch. I have not verified this, but could it be related to https://github.com/stack-of-tasks/pinocchio/pull/1442?

proyan, 2021-06-07:
Hi @nielsvd 
Could you provide an example of 
1) your test with rbdl, 
2) your test with the current master branch, and 
3) your test with the devel branch

It would be helpful if you can show us the code that you are using, and the specific values where you see (if any) divergence occuring.

jcarpent, 2021-06-07:
@nielsvd Yes, you're correct. I've fixed the bug in the devel branch.
Do you need a new release?

nielsvd, 2021-06-07:
@jcarpent , a new release would be great.

nielsvd, 2021-06-07:
@proyan , thanks for your reply. I was actually working on such an example. But as I realized that in `devel` the problem did not exist anymore I have spared some of my time by not finishing it.

jcarpent, 2021-06-08:
Solved in Pinocchio 2.6.1

closed 2021-06-08 by jcarpent