stack-of-tasks/pinocchio #1461: buildReducedModel issue with fixed joint between moving joints

LucasJoseph, 2021-06-15:
Hello,

I noticed an issue with the inertial properties of a `buildReducedModel` when there is a fixed joint located between moving joints. For example, in the panda.urdf file, the gripper has a joint which is fixed to the end-effector (`panda_hand_joint` ) and then two prismatic joints for the fingers (`panda_finger_joint1` and `panda_finger_joint2`). 

I used the code proposed in the [build reduced model example](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_b-examples_o-reduced-model.html) to compute a reduced model of this panda robot. I linked with this issue the code as well as a panda.urdf file with the inertial properties filled [example.zip](https://github.com/stack-of-tasks/pinocchio/files/6658380/example.zip).

If I lock the `panda_finger_joint1` and `panda_finger_joint2` and compare the total mass and the gravity vector for both the original model and the reduce one, I obtain the following values.

```
original model computeTotalMass 16.7862
original model data.g 8.88178e-16     18.6179    -5.31438     8.11598     1.78405      1.7934  -0.0174402     0.22672    -0.22672

reduced model computeTotalMass 17.4423
reduced_model model data.g 8.88178e-16     20.8556      -5.766     8.86355     2.54007     2.52341  -0.0611838
```

It seems that the fixed joint inertial properties are counted twice.

If I set the fixed `panda_hand_joint` to a prismatic then both model are equal (as it should be)
``` 
original model computeTotalMass 16.7862
original model data.g -8.88178e-16       18.455     -5.30146      8.08369      1.67807      1.71499     0.130399     -4.82781      0.22672     -0.22672

reduced_model computeTotalMass 16.7863
reduced_model model data.g 1.77636e-15     18.4553    -5.30154     8.08382     1.67817     1.71508    0.130399    -4.82781
```

jcarpent, 2021-06-15:
Thanks for the report. Which version of Pinocchio are you using. I think the bug has been fixed since Pinocchio 2.6.1

LucasJoseph, 2021-06-15:
I was using the ros version so 2.6.0. I tested it with 2.6.1 and it's indeed working properly. Sorry for the false report.

closed 2021-06-15 by LucasJoseph

wxmerkt, 2021-06-16:
> I was using the ros version so 2.6.0. I tested it with 2.6.1 and it's indeed working properly. Sorry for the false report.

@LucasJoseph on which distro was this? Releases for 2.6.1 are out for both Melodic and Noetic and they will be included in the next sync (you could already access them via `testing`)

LucasJoseph, 2021-06-16:
I'm on Ubuntu 20.04 with a ros noetic version. I install pinocchio through `sudo apt install ros-noetic-pinocchio`

If I do 
```
> rosversion pinocchio
2.6.0
```

wxmerkt, 2021-06-16:
ROS uses a release and a staging server for the packages - i.e. packages that have been released will be available on the staging server for some time to be tested with other packages in the same sync - this used to be called `ros-shadow-fixed` and is now called `ros-testing`. Periodically - approximately every two weeks - all stable packages from staging will be pushed onto the release repository - which you can then access via `apt-get upgrade`. More information is [here](https://wiki.ros.org/TestingRepository).

v2.6.1 is currently on staging and will be available to update during the next sync. To access staging packages 'early' (at the risk of breaking changes), you can adjust your sources file to `ros-testing` rather than `ros`.

LucasJoseph, 2021-06-16:
Interesting, thank you for the information!