stack-of-tasks/pinocchio #1463: Loading URDFs in catkin devel mode

jmirabel, 2021-06-28:
Hi,

when loading URDF model in Pinocchio, one can use `os.env["ROS_PACKAGE_PATH"].split(':')` as list of directories for loading meshes. This however does not work when using catkin devel mode because the directory tree layout is not the same.
I don't think this is Pinocchio's fault but it makes loading URDF in this context slightly painful (in an automated way).

Has anyone a neat solution for this ?

olivier-stasse, 2021-06-28:
The main problem is that all Gepetto packages assume a clean install (i.e. everything is located in the install directory).
You can use:
`catkin config --install`
to have catkin forcing the installation for every package.

I found this to be the easiest way to avoid the devel mess.

jmirabel, 2021-06-28:
Yes, this is a workaround I am aware of. Personally, I always use this mode.

I was wondering whether there could be a easy solution. I personally do not see any. If nothing comes out in the 2-3 coming days, I'll close this.

I haven't checked the code but I guess packages are found by recursively looking for directories with a valid `package.xml` inside (BTW, if you want catkin to run in a infinite loop, add a loop with a symlink in your install folder). The only solution to this problem I can think of is to mimic this behavior. I think this shouldn't be done.

olivier-stasse, 2021-06-28:
Ok i got our point. You want this to work even for people using the devel way of doing thing.
I have a none-technical answer to this: devel does not exist anymore in ros2 with colcon.
I am not sure it is worthwhile to find a solution for this specific case in the long term.

nim65s, 2021-06-28:
Hi @jmirabel,

I tried the following:
```bash
apt install -qqy ros-noetic-pinocchio
source /opt/ros/noetic/setup.bash
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src
git clone --recursive https://github.com/gepetto/example-robot-data
cd ..
catkin_make_isolated -DPYTHON_EXECUTABLE=/usr/bin/python3
source devel_isolated/setup.bash
```

And after that, in a python3 shell:
```python
>>> import example_robot_data
>>> example_robot_data.load_full('talos')
(<pinocchio.robot_wrapper.RobotWrapper object at 0x7ff853bbb040>, array([ 0.00000e+00,  0.00000e+00,  1.01927e+00,  0.00000e+00,
        0.00000e+00,  0.00000e+00,  1.00000e+00,  0.00000e+00,
        0.00000e+00, -4.11354e-01,  8.59395e-01, -4.48041e-01,
       -1.70800e-03,  0.00000e+00,  0.00000e+00, -4.11354e-01,
        8.59395e-01, -4.48041e-01, -1.70800e-03,  0.00000e+00,
        6.76100e-03,  2.58470e-01,  1.73046e-01, -2.00000e-04,
       -5.25366e-01,  0.00000e+00,  0.00000e+00,  1.00000e-01,
        0.00000e+00, -2.58470e-01, -1.73046e-01,  2.00000e-04,
       -5.25366e-01,  0.00000e+00,  0.00000e+00,  1.00000e-01,
        0.00000e+00,  0.00000e+00,  0.00000e+00]), '/root/catkin_ws/devel_isolated/example-robot-data/share/example-robot-data/robots/talos_data/robots/talos_reduced.urdf', '/root/catkin_ws/devel_isolated/example-robot-data/share/example-robot-data/robots/talos_data/srdf/talos.srdf')
```

So I guess in this case this is working well in this case. Is this helping ?

Maybe this is because the layout of example-robot-data is done as expected, or because I'm using `catkin_make_isolated` (the [doc]( http://wiki.ros.org/catkin/Tutorials/create_a_workspace) says we should use `catkin_make`,  but `catkin_make`says that my package is not a catkin package, and therefore I should use `catkin_make_isolated`…)

cmastalli, 2021-06-28:
Now the standard is 'catkin tools'. @nim65s -- it would be helpful if you can try it again with 'catkin tools'.

nim65s, 2021-06-28:
I can get the same result with catkin-tools:

```bash
apt install -qqy ros-noetic-pinocchio python-is-python3 python3-catkin-tools python3-osrf-pycommon
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src/
git clone --recursive https://github.com/gepetto/example-robot-data
cd ..
catkin init
catkin build
source devel/setup.bash
```

jmirabel, 2021-06-29:
This is not a correct example as this package does not use ROS_PACKAGE_PATH.

wxmerkt, 2021-07-01:
I don't see this issue with devel spaces and Pinocchio when loading URDFs directly across many different robot repositories.

@jmirabel which robot are you seeing the issue with (e.g. description repo) and how are you loading it?

The one case where I see failure is if a xacro is compiled in a devel space with incorrect arguments - here the order of arguments to `xacro_add_files` is key to make it work (i.e. it needs to also create the URDF in the source tree, and not the symlinked devel tree only).

jmirabel, 2021-07-01:
I'll try to provide a minimal reproducible example.

jmirabel, 2021-07-01:
Using ur_description package, I get the error.
Start from docker image `ros:melodic-ros-base-bionic`, setting up a catkin devel workspace, ROS_PACKAGE_PATH is `/ws/src/universal_robot/ur_description:/opt/ros/melodic/share`.
Clearly, pinocchio's strategy to find mesh from `package://ur_description/meshes/...` fails. It will look for `/ws/src/universal_robot/ur_description/ur_description/meshes/...`. Pinocchio would expect ROS_PACKAGE_PATH to be `/ws/src/universal_robot:/opt/ros/melodic/share`.

wxmerkt, 2021-07-01:
Gotcha, thanks for providing the details. Indeed, this is a correct observation: the `ROS_PACKAGE_PATH` works differently for devel/install spaces and the solution is likely to go relative to where the `package.xml` is - or to also search the directory one above (`/..`) the directories returned in `rosPaths`. This should catch the majority of cases.

@nim65s `ros-[melodic/noetic]-pinocchio` are not built with HPP-FCL currently as thus I don't think this reproduction case works as intended. I just create another one with Docker and it suffers from the same fate.

jmirabel, 2021-07-01:
I had to use robotpkg in the docker to use pinocchio with hpp-fcl. Not hard to do but add some steps.

wxmerkt, 2021-07-11:
Solved by #1467

closed 2021-07-11 by wxmerkt