stack-of-tasks/pinocchio #1464: compiling error in ros

ddliu365, 2021-07-01:
Hi,

I met a compiling error combing pinocchio in ROS.

```
[100%] Linking CXX executable /home/jack/catkin_ws/devel/lib/humanoid_core/humanoid_core
/usr/bin/ld: /home/jack/catkin_ws/devel/lib/libcore_modules.so: undefined reference to `pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&)'
```

In my CMakeLists.txt, I configured like this:

```
add_executable(${PROJECT_NAME} src/core.cpp)

add_dependencies(${PROJECT_NAME} humanoid_core_gencpp)

target_link_libraries(${PROJECT_NAME} ${catkin_LIBRARIES} ${Eigen3_LIBRARIES} ${PINOCCHIO_LIBRARIES} core_modules)
target_compile_options(${PROJECT_NAME} PUBLIC ${pinocchio_CFLAGS_OTHER})
```

It actually worked in ROS kinetic, but when we migrated everything to Ubuntu 20.04(ROS noetic), it shows up.

I am not sure if the version will make any difference in the compiling, but it seems this error is caused by messing up the compiling option.

Does anyone can give an suggestion to solve it?

jcarpent, 2021-07-01:
Before answering your issue, I think important to clarify some points with you @ddliu365.
It is bit frustrating from on our side as You always show the same kind of errors, do you know that?
We may take some times to answer questions that are really related to Pinocchio. Here the issue is coming from your side. 
Additionally, it would be nice to cite Pinocchio in your research in addition to Crocoddyl. As far as I know, your are requesting some helps without any acknowledgment for our work up to now.

Quick answer: Here your issue is coming from the fact you are not compiling with C++11 on.

closed 2021-07-01 by jcarpent

ddliu365, 2021-07-01:
Actually I added C++ 11 to the CMakeLists.txt. Sorry, I didn't show a full file. 
```
cmake_minimum_required(VERSION 3.0.2)
project(humanoid_core)

## Compile as C++11, supported in ROS Kinetic and newer
add_compile_options(-std=c++11)
add_definitions("-DBOOST_MPL_LIMIT_LIST_SIZE=30")

find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs
  roslib
  message_generation
)

find_package(Eigen3 REQUIRED)

find_package(pinocchio REQUIRED)

add_service_files(
  FILES
  core_srv.srv
)

## Generate added messages and services with any dependencies listed here
generate_messages(
  DEPENDENCIES
  std_msgs
)

catkin_package(
 INCLUDE_DIRS include
 LIBRARIES humanoid_core
 CATKIN_DEPENDS roscpp rospy std_msgs roslib
 DEPENDS EIGEN3
)

include_directories(
 include
  ${catkin_INCLUDE_DIRS}
  ${EIGEN3_INCLUDE_DIRS}
  ${PINOCCHIO_INCLUDE_DIRS}
)

## Declare a C++ library
add_library(core_modules
  src/estimation_module.cpp
  src/data_module.cpp
  src/invkin_module.cpp
  src/pid.cpp
)

add_executable(${PROJECT_NAME} src/core.cpp)

add_dependencies(${PROJECT_NAME} humanoid_core_gencpp)

## Specify libraries to link a library or executable target against
target_link_libraries(${PROJECT_NAME} ${catkin_LIBRARIES} ${Eigen3_LIBRARIES} ${PINOCCHIO_LIBRARIES} core_modules)
target_compile_options(${PROJECT_NAME} PUBLIC ${pinocchio_CFLAGS_OTHER})
```

Sorry for my disturb. You always provide the quick response and we definitely acknowledge your work in the future! Maybe my question is stupid from your point, but your answer is helpful to me.

wxmerkt, 2021-07-01:
Change `${PINOCCHIO_LIBRARIES}` to `pinocchio::pinocchio` in `target_link_libraries`

ddliu365, 2021-07-01:
Thanks for your reply!

```
target_link_libraries(${PROJECT_NAME} pinocchio::pinocchio ${catkin_LIBRARIES} ${Eigen3_LIBRARIES}  core_modules)
```
I got the same error

ddliu365, 2021-07-01:
My problem is solved by tuning the order
I moved self-defined library, which depends on `pinocchio`,  from the right most position to the left. It worked but not sure why.

```
target_link_libraries(${PROJECT_NAME}  core_modules ${catkin_LIBRARIES} ${Eigen3_LIBRARIES}  pinocchio::pinocchio)
```
Thanks for your help! @wxmerkt @jcarpent

wxmerkt, 2021-07-02:
> I moved self-defined library, which depends on pinocchio, from the right most position to the left. It worked but not sure why.

I'd estimate that a `target_link_libraries(core_modules pinocchio::pinocchio [libraries this depends on])` is missing

jcarpent, 2021-07-02:
Do agree with @wxmerkt.