stack-of-tasks/pinocchio #1468: [c++] SE3 * Vector3

jmirabel, 2021-07-09:
Hi,

To multiply a SE3 object with a Eigen::Vector3, one needs to do `M.act(p)` because `M * p` fails.

Is there a reason for not accepting `M * p` ?

jcarpent, 2021-07-09:
Currently, we use `M * object` only on Spatial quantities.
In the context of Eigen types, we should check the size of the objects at compile time to be sure that we have indeed a 3d vector. Not done currently, and don't think this will be helpful.

jmirabel, 2021-07-09:
I agree we have to check the size but it can be done at runtime if not possible at compile time (i.e. dynamic size vectors).

It seems there is nothing blocking. It's only a matter of developer time ?

jmirabel, 2021-07-09:
API-wise, I repeatedly have the compiler telling me the operator* could not be found so, at least for me, it would be useful.

jcarpent, 2021-07-09:
But this means that you will have a loss of performance by checking at runtime.
For me, I'm not in favor, I'm not against, but when I extended the Spatial classes, the `operator *` was dedicated to Spatial quantities, as we may imagine having `M * Matrix6x` too, but this has no real meaning, as we don't know if Matrix6x is a Motion or Force set.
I would be carefully before introducing this multiply operator for 3d vectors.

jmirabel, 2021-07-09:
I have my answer now.
`SE3 * Matrix` can have several meaning. This is sort of blocking. Or at least, it requires the API to be well thought.
Thanks.

closed 2021-07-09 by jcarpent