stack-of-tasks/pinocchio #1471: Talos robot not correctly loaded in the meshcat-viewer example

GiulioRomualdi, 2021-07-16:
Hi all, first of all, thank you for the effort to develop and maintain `pinocchio`. 

I was trying to visualize the Talos robot following [this example](https://github.com/stack-of-tasks/pinocchio/blob/b134b25f1409f5bf036105b996da2d29c1a66a12/examples/meshcat-viewer.py). However I noticed that model is not correctly loaded in meshcat.

Indeed there seems that the right shoulder and the legs are not correctly attached to the robot body. 

| Real Talos | Visualized Talos |
| :----:|:----:|
| [image] | [image] |

I installed `meshcat` and `pinocchio` with conda in a clean `Ubuntu 20.04` distro.

For reference, my packages are:
```
conda list
# packages in environment at /home/gromualdi/miniforge3/envs/pinocchio-env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                      1_llvm    conda-forge
assimp                    5.0.1                hedfc422_5    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
boost                     1.74.0           py39h5472131_3    conda-forge
boost-cpp                 1.74.0               h312852a_4    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
ca-certificates           2021.5.30            ha878542_0    conda-forge
certifi                   2021.5.30        py39hf3d152e_0    conda-forge
console_bridge            1.0.1                h4bd325d_0    conda-forge
decorator                 5.0.9              pyhd8ed1ab_0    conda-forge
eigen                     3.3.9                h4bd325d_1    conda-forge
eigenpy                   2.6.4            py39h5aed9d1_0    conda-forge
hpp-fcl                   1.7.4            py39hbcdfc36_0    conda-forge
icu                       68.1                 h58526e2_0    conda-forge
ipython                   7.25.0           py39hef51801_1    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
jedi                      0.18.0           py39hf3d152e_2    conda-forge
ld_impl_linux-64          2.36.1               hea4e1c9_1    conda-forge
libblas                   3.9.0                9_openblas    conda-forge
libcblas                  3.9.0                9_openblas    conda-forge
libffi                    3.3                  h58526e2_2    conda-forge
libgcc-ng                 9.3.0               h2828fa1_19    conda-forge
libgfortran-ng            9.3.0               hff62375_19    conda-forge
libgfortran5              9.3.0               hff62375_19    conda-forge
liblapack                 3.9.0                9_openblas    conda-forge
libopenblas               0.3.15          pthreads_h8fe5266_1    conda-forge
libsodium                 1.0.18               h36c2ea0_1    conda-forge
libstdcxx-ng              9.3.0               h6de172a_19    conda-forge
llvm-openmp               12.0.1               h4bd325d_0    conda-forge
lz4-c                     1.9.3                h9c3ff4c_0    conda-forge
matplotlib-inline         0.1.2              pyhd8ed1ab_2    conda-forge
meshcat-python            0.1.1              pyhd8ed1ab_0    conda-forge
ncurses                   6.2                  h58526e2_4    conda-forge
numpy                     1.21.0           py39hdbf815f_0    conda-forge
octomap                   1.9.7                h4bd325d_0    conda-forge
openssl                   1.1.1k               h7f98852_0    conda-forge
parso                     0.8.2              pyhd8ed1ab_0    conda-forge
pexpect                   4.8.0              pyh9f0ad1d_2    conda-forge
pickleshare               0.7.5                   py_1003    conda-forge
pinocchio                 2.6.2            py39hbcdfc36_0    conda-forge
pip                       21.1.3             pyhd8ed1ab_0    conda-forge
prompt-toolkit            3.0.19             pyha770c72_0    conda-forge
ptyprocess                0.7.0              pyhd3deb0d_0    conda-forge
pygments                  2.9.0              pyhd8ed1ab_0    conda-forge
pyngrok                   5.0.5              pyhd8ed1ab_0    conda-forge
python                    3.9.6           h49503c6_1_cpython    conda-forge
python_abi                3.9                      2_cp39    conda-forge
pyyaml                    5.4.1            py39h3811e60_0    conda-forge
pyzmq                     22.1.0           py39h37b5a0c_0    conda-forge
readline                  8.1                  h46c0cb4_0    conda-forge
setuptools                49.6.0           py39hf3d152e_3    conda-forge
sqlite                    3.36.0               h9cd32fc_0    conda-forge
tinyxml                   2.6.2                h4bd325d_2    conda-forge
tk                        8.6.10               h21135ba_1    conda-forge
tornado                   6.1              py39h3811e60_1    conda-forge
traitlets                 5.0.5                      py_0    conda-forge
tzdata                    2021a                he74cb21_1    conda-forge
u-msgpack-python          2.7.1              pyh9f0ad1d_0    conda-forge
urdfdom                   2.3.5                h4bd325d_0    conda-forge
urdfdom_headers           1.0.5                h4bd325d_2    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
wheel                     0.36.2             pyhd3deb0d_0    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h516909a_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_0    conda-forge
zlib                      1.2.11            h516909a_1010    conda-forge
zstd                      1.5.0                ha95c52a_0    conda-forge
```

jcarpent, 2021-07-16:
Thanks @GiulioRomualdi for raising this issue and reporting for this bug.
I think this is due to some slight but important changes in the way we parse frames in the URDF parser and which has been introduced in Pinocchio 2.6.0.
You can use 2.5.6 in meantime, and I will provide a fix later this day.

GiulioRomualdi, 2021-07-16:
Hi @jcarpent. Thank you! With pinocchio 2.5.6 the problem disappears

jcarpent, 2021-07-16:
@GiulioRomualdi I've fixed the bug and in fact it was only related to Meshcat support of Pinocchio.
Please cherry-pick this commit 24dbede to solve locally your issue.

I will provide a release soon to fix it at least on conda.
Thanks again for the quick report.

jcarpent, 2021-07-16:
@GiulioRomualdi Could you confirm that fixes your issue in robotology/icub-models#97?

GiulioRomualdi, 2021-07-16:
> @GiulioRomualdi I've fixed the bug and in fact it was only related to Meshcat support of Pinocchio.
> Please cherry-pick this commit [...]

https://github.com/stack-of-tasks/pinocchio/commit/24dbede81e884dc0be2bb05ecd1d3f7171cba982 solves the problem with Talos and Pinocchio 2.6.2. However, when I try to load the iCub (robotology/icub-models#97) the meshes are not scaled  

[image]

without 24dbede the meshes are scaled even if the robot is not correctly loaded

GiulioRomualdi, 2021-07-16:
Adding the following lines after

```python
    q0 = pin.neutral(model)
    viz.display(q0)
```
`viz.loadViewerModel()`  [here](https://github.com/GiulioRomualdi/study-icub-meshcat/blob/83191c001d2e384ee7d50db7de97013481f5a1b2/visualize-icub.py#L45) 
display the robot scaled and in the neutral configuration.

Looking in the code seems the meshes are scaled only when `display` is called (e.g. https://github.com/stack-of-tasks/pinocchio/pull/1472/commits/50f49cc7fc3ddd110f4bd09d381b9afb081e7ffa). Is it wanted?

jcarpent, 2021-07-16:
I've just update #1472 to also scale the geometries when loading the meshes.
Yet, it seems that there is a bug within MeshCat to use the scaling information when changing the placement (via `set_transform`).

jcarpent, 2021-07-16:
Small remark: according to https://github.com/rdeits/meshcat/blob/1ea52a649c0c8df800587b54d0910ef2868f8de5/src/index.js#L357-L361, it seems that Meshcat erases the previous scaling.

jcarpent, 2021-07-16:
Solved by #1472

closed 2021-07-16 by jcarpent

traversaro, 2021-07-17:
Thanks a lot @jcarpent for the fix!

akhilsathuluri, 2022-06-18:
Hello, I am trying to reproduce the exact scenario as above. The iCub model loads to the viewer by,
```
viz.initViewer(loadModel=True)
```
as shown below,
[image]

However, when I modify the pose of the robot (or even in neutral) by setting certain angles and upon visualising it using,
 ```
q0 = pin.neutral(robot.model())
viz.display(q0)
```
the robot model completely disappears from the `meshcat` viewer. I suspect this has to do with the scaling too. 

My setup: `Ubuntu 20.04`, `pinocchio-2.6.0`

jcarpent, 2022-06-18:
Could you use a more recent version of Pinocchio?

akhilsathuluri, 2022-06-18:
Thanks for a super quick reply. I had a bunch of dependency issues between `tsid` and `pinocchio` versions. But finally `pinocchio=2.6.3` with `tsid=1.6.1` fixed the issue.

rocketman123456, 2024-12-09:
when i use pinocchio 3.3.0 on macos, i still get the problem of talos model incorrect loading with the meshcat talos example. what can i do to fix it?
conda list:
```
# packages in environment at /Users/rocketsky/miniconda3/envs/pinocchio:
#
# Name                    Version                   Build  Channel
absl-py                   2.1.0                    pypi_0    pypi
ampl-asl                  1.0.0                h286801f_2    conda-forge
assimp                    5.4.3                ha9c0b8d_0    conda-forge
asttokens                 3.0.0                    pypi_0    pypi
bzip2                     1.0.8                h80987f9_6    defaults
ca-certificates           2024.11.26           hca03da5_0    defaults
casadi                    3.6.7           py310h3df02a9_0    conda-forge
cli11                     2.4.2                h286801f_0    conda-forge
coal                      3.0.0                hd8ed1ab_5    conda-forge
coal-python               3.0.0           py310hd51063e_5    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
crocoddyl                 2.1.0           py310h7dd35b7_8    conda-forge
decorator                 5.1.1                    pypi_0    pypi
eigen                     3.4.0                h1995070_0    conda-forge
eigenpy                   3.10.1          np20py310h377b2b4_0    conda-forge
etils                     1.11.0                   pypi_0    pypi
example-robot-data        4.1.0              pyh6b898a9_5    conda-forge
exceptiongroup            1.2.2                    pypi_0    pypi
executing                 2.1.0                    pypi_0    pypi
fsspec                    2024.10.0                pypi_0    pypi
gettext                   0.22.5               h8414b35_3    conda-forge
gettext-tools             0.22.5               h8414b35_3    conda-forge
glfw                      2.8.0                    pypi_0    pypi
gmp                       6.3.0                h7bae524_2    conda-forge
hpp-fcl                   3.0.0              pyhd3e5650_4    conda-forge
icu                       75.1                 hfee45f7_0    conda-forge
importlib-resources       6.4.5                    pypi_0    pypi
ipopt                     3.14.16             h3e4dc2c_11    conda-forge
ipython                   8.30.0                   pypi_0    pypi
jedi                      0.19.2                   pypi_0    pypi
libasprintf               0.22.5               h8414b35_3    conda-forge
libasprintf-devel         0.22.5               h8414b35_3    conda-forge
libblas                   3.9.0           25_osxarm64_openblas    conda-forge
libboost                  1.86.0               hc9fb7c5_3    conda-forge
libboost-devel            1.86.0               hf450f58_3    conda-forge
libboost-headers          1.86.0               hce30654_3    conda-forge
libboost-python           1.86.0          py310hf1156d2_3    conda-forge
libboost-python-devel     1.86.0          py310h24597f5_3    conda-forge
libcblas                  3.9.0           25_osxarm64_openblas    conda-forge
libcoal                   3.0.0                hf5bebb4_5    conda-forge
libcxx                    19.1.5               ha82da77_0    conda-forge
libffi                    3.4.4                hca03da5_1    defaults
libgettextpo              0.22.5               h8414b35_3    conda-forge
libgettextpo-devel        0.22.5               h8414b35_3    conda-forge
libgfortran               5.0.0           13_2_0_hd922786_3    conda-forge
libgfortran5              13.2.0               hf226fd6_3    conda-forge
libgz-cmake3              3.5.3                h00cdb27_0    conda-forge
libgz-math7               7.5.1                h5833ebf_2    conda-forge
libgz-tools2              2.0.1                hce4b3f6_0    conda-forge
libgz-utils2              2.2.0                hf9b8971_0    conda-forge
libiconv                  1.17                 h0d3ecfb_2    conda-forge
libintl                   0.22.5               h8414b35_3    conda-forge
libintl-devel             0.22.5               h8414b35_3    conda-forge
liblapack                 3.9.0           25_osxarm64_openblas    conda-forge
liblzma                   5.6.3                h39f12f2_1    conda-forge
libopenblas               0.3.28          openmp_hf332438_1    conda-forge
libosqp                   0.6.3                h5833ebf_1    conda-forge
libqdldl                  0.1.7                hb7217d7_0    conda-forge
libscotch                 7.0.5                hc2b2845_4    conda-forge
libsdformat14             14.5.0               he19963e_0    conda-forge
libsqlite                 3.45.2               h091b4b1_0    conda-forge
libzlib                   1.3.1                h8359307_2    conda-forge
llvm-openmp               19.1.5               hdb05f8b_0    conda-forge
matplotlib-inline         0.1.7                    pypi_0    pypi
[... 26 lines cut ...]
qhull-static              2020.2               h420ef59_5    conda-forge
readline                  8.2                  h1a28f6b_0    defaults
ruby                      3.2.2                ha6ee62f_1    conda-forge
scipy                     1.14.1          py310hed58976_2    conda-forge
setuptools                75.1.0          py310hca03da5_0    defaults
simde                     0.8.2                h7b3277c_0    conda-forge
sqlite                    3.45.2               hf2abe2d_0    conda-forge
[... cut ...]
urdfdom_headers           1.1.2                h7b3277c_0    conda-forge
wcwidth                   0.2.13                   pypi_0    pypi
wheel                     0.44.0          py310hca03da5_0    defaults
xz                        5.4.6                h80987f9_1    defaults
yaml                      0.2.5                h3422bc3_2    conda-forge
zipp                      3.21.0                   pypi_0    pypi
zlib                      1.3.1                h8359307_2    conda-forge
zstd                      1.5.6                hb46c0d2_0    conda-forge
```