stack-of-tasks/pinocchio #1473: Analytical Derivative of Centroidal Momentum

jp-sleiman, 2021-07-26:
Hello, 

Thanks for making pinocchio available!

I am currently comparing the analytical derivative of the centroidal momentum with respect to the configuration vector (**dh_dq** extracted from `computeCentroidalDynamicsDerivatives`) to that given by a CppAD implementation. I noticed that there is a mismatch between the two results.
What was mostly confusing to me is that the analytical derivative with respect to the floating-base position was different from zero, which would contradict the fact that the centroidal momentum matrix and centroidal momentum do not vary when changing the base position. The only explanation that comes to mind is that **dh_dq** is the derivative of the centroidal momentum expressed in the world frame instead of the centroidal frame. Is this correct, or am I missing something? Also if this is true, is there a way to get **dh_dq** in the centroidal frame?

Thanks a lot!
Cheers,
Jean-Pierre

jcarpent, 2021-07-26:
Dear @jp-sleiman,

Indeed, the current derivatives for the centroildal dynamics are expressed in the frame centered at the CoM and aligned with the WORLD coordinate axes. This is made on purpose, as there is no clear notion of centroidal orientation.
We may choose to select by default the first joint, which should be a free_flyer for the LOCAL notion.
Is it what you need? Why you could not use the current output?

jcarpent, 2021-07-26:
To be more precise, we may imagine to have the two versions:
- LOCAL: dh_dq expressed in the frame aligned with the first joint frame
- LOCAL_WORLD_ALIGNED: the current output

jp-sleiman, 2021-07-26:
Hi @jcarpent, thanks for the quick reply!

Yes, that is indeed what I need: the centroidal dynamics expressed in the frame centered at the CoM and aligned with the WORLD axes.

But in this case shouldn't I expect the derivative of the centroidal momentum with respect to the free_flyer position to be zero? What confused me is that, unlike the CppAD result, it wasn't zero, and the CppAD result seems more intuitive to me since the value of the centroidal momentum should not vary with the free_flyer position. So I was wondering if there is an error in the analytical derivatives computation, or if there is something missing in my explanation.

jcarpent, 2021-07-26:
I don't know how you proceed with CppAD, but in your case, the part due to the free-flyer is due to the translation of the CoM when changing q. Is it clear?
You can cancel this term. This is what you need?

jp-sleiman, 2021-07-26:
It is clear that the translation of the free-flyer results in a translation of the CoM, but it is not clear to me why this would cause a change in the centroidal momentum if all other quantities (joint angles, base orientation) are fixed. To further investigate this, I also computed, using pinocchio, the centroidal momenta **h1** and **h2** for configurations **q1** and **q2** where the difference between the configurations is only in the free-flyer position. **h1** and **h2** match in this case. Which is why I also expected the three columns of **dh_dq** corresponding to the free-flyer position to be zero.

jcarpent, 2021-07-26:
I've found the bug and it is related to the fact that I've missed a term when computing the partial w.r.t q. 
I will provide a fix today.

jp-sleiman, 2021-07-26:
Okay great, thanks a lot for your help!

closed 2021-07-26 by jp-sleiman