stack-of-tasks/pinocchio #1475: Trashed .dae meshes?

danzimmerman, 2021-07-26:
I'm having trouble with `conda`-installed MeshCat and Pinocchio trying to visualize a [UR5 from `example-robot-data`](https://github.com/Gepetto/example-robot-data) (gray) even though each individual file renders well in MeshCat (red)

[image]

The code for this example is collapsed below:
<details>

```python
import example_robot_data
import pinocchio as pin
import numpy as np
import meshcat

robot = example_robot_data.load('ur5')

# Currently, MeshCat is not able to retrieve the scaling from DAE files. Set it manually.
for geom in robot.visual_model.geometryObjects:
    s = geom.meshScale 
    s *= 1.0
    geom.meshScale = s
robot.setVisualizer(pin.visualize.MeshcatVisualizer(robot.model, robot.collision_model, robot.visual_model))
robot.initViewer()
robot.loadViewerModel('ur5_model')
robot.display(robot.q0)

# -- extract the concrete .dae paths from the text repr of robot.visual_model --
s = repr(robot.visual_model)
meshes = {} #key, value like wrist2:path/to/wrist2.dae
blocks = s.split('Absolute path to mesh file:')[1:]
for block in blocks:
    meshpath = block.split('Scale')[0].strip()
    meshkey = meshpath.split('/')[-1].replace('.dae', '')
    meshes[meshkey] = meshpath

# -- add wrist2.dae as a comparison 
material = meshcat.geometry.MeshPhongMaterial()
material.color = 0x802317
dae_file = meshes['wrist2']
robot.viz.viewer['wrist2_comparison'].set_object(meshcat.geometry.DaeMeshGeometry.from_file(dae_file), material)
print(f'Adding {dae_file}')
T = np.array([[-1.,  0.,  0.,  0.8],
              [ 0.,  1.,  0.,  0.3],
              [-0.,  0., -1.,  0.1],
              [ 0.,  0.,  0.,  1.]])

robot.viz.viewer['wrist2_comparison'].set_transform(T)
```

</details>

After I developed this example, I found the [`meshcat-viewer-dae.py` example](https://github.com/stack-of-tasks/pinocchio/blob/master/examples/meshcat-viewer-dae.py) and the suggestion that `robot_wrapper` is on its way out as a recommended method, but quickly following a similar track to `meshcat-viewer-dae.py` results in mangled DAE files as well.

Any suggestions appreciated!

**Operating System:**
```
Edition	Windows 11 Home
Version	21H2
Installed on	‎7/‎1/‎2021
OS build	22000.100
```
**Selected Output of `conda list`**:
```
pinocchio                 2.6.2            py38hb4864f1_0    conda-forge
meshcat-python            0.1.1              pyhd8ed1ab_0    conda-forge
example-robot-data        3.10.0           py38haa244fe_0    conda-forge
```

**Full Output of `conda list`**:
<details>
[... 634 lines cut ...]
winpty                    0.4.3                         4    conda-forge
wstool                    0.1.17                   pypi_0    pypi
xerces-c                  3.2.3                h0e60522_2    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_1    conda-forge
xorg-libxpm               3.5.13               hcd874cb_0    conda-forge
xorg-libxt                1.2.1                hcd874cb_2    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1002    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zeromq                    4.3.4                h0e60522_0    conda-forge
zfp                       0.5.5                h0e60522_5    conda-forge
zipp                      3.5.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.5.0                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```
</details>

jcarpent, 2021-07-27:
I've tried on my side, and it seems that the issue is comming from MeshCat and your browser.
Indeed, depending on the tree.js version present in your browser, the capacity of loading DAE files might be affected.

closed 2021-07-27 by jcarpent

danzimmerman, 2021-07-27:
Thanks, I'm having the same problem on Chrome, Microsoft Edge, and Firefox across a couple different machines, but with the same version of MeshCat. I'll look in the direction of MeshCat + three.js

jcarpent, 2021-07-27:
On my side, this is working well on Firefox 90.0.2 but not with Safari (on OSX).

danzimmerman, 2021-07-27:
The `conda`-installed version of Pinocchio I had was not the latest, v `2.6.2` even though `2.9.1` is [available on `conda-forge`](https://anaconda.org/conda-forge/pinocchio). 

It's calling `loadBVH()` [here](https://github.com/stack-of-tasks/pinocchio/blob/v2.6.2/bindings/python/pinocchio/visualize/meshcat_visualizer.py#L171) and if I force it to call `self.loadMesh()` instead, it works fine:

[image]

I think I didn't pick up `2.9.1` (which fixes this with [the `isMesh()` test](https://github.com/stack-of-tasks/pinocchio/blob/v2.9.1/bindings/python/pinocchio/visualize/meshcat_visualizer.py#L184)) perhaps because all the `2.9.1` builds are labeled `broken`:

https://anaconda.org/conda-forge/pinocchio/files

There also seems to be a conflict with `example-robot-data`:

```
mamba install pinocchio==2.9.1 example-robot-data -c conda-forge/label/broken --dry-run
```

it fails with 

```
Problem: package example-robot-data-3.10.0-py38haa244fe_0 requires pinocchio >=2.6.1,<2.7.0a0, but none of the providers can be installed
```

If I uninstall `example-robot-data` I can install `2.9.1` with

```
mamba install pinocchio==2.9.1 example-robot-data -c conda-forge/label/broken
```
and it works fine to pull the `.dae` files out of `ur_description` for my original application, so problem solved there.