stack-of-tasks/pinocchio #1477: RobotWrapper(URDF), ValueError: Mesh file could not be found.

beta1scat, 2021-07-28:
System: Ubuntu 18.04
ROS: Melodic
Pinocchio: robotpkg-py36-pinocchio

I am learning Pinocchio tutorials: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_1-directgeom.html

I have cloned package "example-robot-data" from https://github.com/Gepetto/example-robot-data in my ROS catkin workspace and build successed. But when I want to use URDF file in  ROS package "example-robot-data" with my python scripts, an error occered:
`ValueError: Mesh package://example-robot-data/robots/solo_description/meshes/obj/with_foot/solo_body.obj could not be found.
`

My scripts is :
```
from pinocchio.robot_wrapper import RobotWrapper

URDF = '/home/niu/catkin_ws/src/example-robot-data/robots/ur_description/urdf/ur5_robot.urdf'
robot = RobotWrapper.BuildFromURDF(URDF)
for name, function in robot.model.__class__.__dict__.items():
    print(' **** %s: %s' % (name, function.__doc__))

```

I have sourced: devel/setup.bash and ROS_PACKAGE_PATH is OK.
My ~/.bashrc file is:
```
export PATH=/opt/openrobots/bin:$PATH
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
export PYTHONPATH=/opt/openrobots/lib/python3.6/site-packages:$PYTHONPATH # Adapt your desired python version here
export CMAKE_PREFIX_PATH=/opt/openrobots:$CMAKE_PREFIX_PATH
source /opt/ros/melodic/setup.bash
source /home/niu/catkin_ws/devel/setup.bash
```

ROS_PACKAGE_PATH:
`ROS_PACKAGE_PATH=/home/niu/catkin_ws/src/example-robot-data:/opt/ros/melodic/share`

beta1scat, 2021-07-28:
When I use 
`model    = pin.buildModelFromUrdf(URDF)`
there is no error.

jcarpent, 2021-07-28:
@proyan Could you help on this?

jcarpent, 2021-07-28:
I think `ROS_PACKAGE_PATH=/home/niu/catkin_ws/src` should make the job.

jcarpent, 2021-07-28:
You can also precise the package dirs as follows:
```
robot = RobotWrapper.BuildFromURDF(URDF,package_dirs=['/home/niu/catkin_ws/src'])
```
I will close as the problem will be solved from mines or @nim65s' suggestions

closed 2021-07-28 by jcarpent

beta1scat, 2021-07-29:
Thank you very much! @jcarpent @nim65s 
All your methods are OK!