stack-of-tasks/pinocchio #1481: Shape error in Snap your fingers

kozakromch, 2021-08-02:
Hi, I try to make an Snap your fingers Practical Exercises, but in file https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/robot__hand_8py_source.html was some shape issues in collision related function. 

I make it work, by reshaping some arrays, but to be honest, i don't know is it right. 

This is the collision function, that can work
```
def collision(self, c2, data=None, oMj1=None, oMj2=None):
        if data is not None:
            oMj1 = data.oMi[self.jointParent]
            oMj2 = data.oMi[c2.jointParent]
        M1 = oMj1 * self.placement
        M2 = oMj2 * c2.placement
        assert (self.isCapsule() and c2.isCapsule())
        l1 = self.length
        r1 = self.radius
        l2 = c2.length
        r2 = c2.radius

        a1 = M1.act(np.array([0, 0, -l1 / 2]).T)
        b1 = M2.act(np.array([0, 0, -l2 / 2]).T)
        a2 = M1.act(np.array([0, 0, +l1 / 2]).T)
        b2 = M2.act(np.array([0, 0, +l2 / 2]).T)
        
        ab = pinv((np.hstack((a1 - a2, b2 - b1))).reshape(3, 2)) * (b2 - a2)

        if (0 <= ab).all() and (ab <= 1).all():
            asat = bsat = False
            pa = a2 + ab[0, 0] * (a1 - a2)
            pb = b2 + ab[1, 0] * (b1 - b2)
        else:
            asat = bsat = True
            i = np.argmin(np.vstack([ab, 1 - ab]))

            pa = a2 if i == 0 else a1
            pb = b2 if i == 1 else b1
            if i == 0 or i == 2:  # fix a to pa, search b
                b = (pinv((b1 - b2).reshape(3, 1)) * (pa - b2))[0, 0]
                if b < 0:
                    pb = b2
                elif b > 1:
                    pb = b1
                else:
                    pb = b2 + b * (b1 - b2)
                    bsat = False
            else:  # fix b
                a = (pinv((a1 - a2).reshape(3, 1)) * (pb - a2))[0, 0]
                if a < 0:
                    pa = a2
                elif a > 1:
                    pa = a1
                else:
                    pa = a2 + a * (a1 - a2)
                    asat = False

        dist = norm(pa - pb) - (r1 + r2)
        if norm(pa - pb) > 1e-3:
            # Compute witness points
            ab = pa - pb
            ab /= norm(ab)
            wa = pa - ab * r1
            wb = pb + ab * r2

            # Compute normal matrix
            x = np.matrix([1., 0, 0]).T
            r1 = cross(ab, x)
            if norm(r1) < 1e-2:
                x = np.matrix([0, 1., 0]).T
            r1 = cross(ab, x)
            r1 /= norm(r1)
            r2 = cross(ab, r1)
            R = np.hstack([r1, r2, ab.reshape(3,1)])

            self.dist = dist
            c2.dist = dist
            self.w = wa
            c2.w = wb
            self.R = R
            c2.R = R

        return dist
``` 

This is the code, that work with new robot_hand (nothing new, just for reproducing)
``` 

from robot_hand import Robot
import gepetto.corbaserver

import pinocchio as pin
import numpy as np
import time
# gepetto.corbaserver.start_server(); time.sleep(5)
robot = Robot()
robot.display(robot.q0)

# Create 10 witness points in the rendering window
for i in range(10):
    robot.viewer.viewer.gui.addCylinder('world/wa%i' % i, .01, .003, [1, 0, 0, 1])
    robot.viewer.viewer.gui.addCylinder('world/wb%i' % i, .01, .003, [1, 0, 0, 1])
    robot.viewer.viewer.gui.setVisibility('world/wa%i' % i, 'OFF')
    robot.viewer.viewer.gui.setVisibility('world/wb%i' % i, 'OFF')
# Add 4 pairs between finger tips and palm
robot.collisionPairs.append([2, 8])
robot.collisionPairs.append([2, 11])
robot.collisionPairs.append([2, 14])
robot.collisionPairs.append([2, 16])
# Compute distance between object 2 and 8, i.e the first collision pair
time.sleep(4)
for i in range (4):
    idx = i
    dist = robot.checkCollision(idx)
    # Display the collision pair by adding two disks at the witness points.
    robot.displayCollision(idx, 0)
    time.sleep(1)
``` 

thank you for such a cool library

jcarpent, 2021-08-02:
Thanks @kozakromch for reporting.
This tutorial is now a bit outdated, as people should now rely on hpp-fcl lib for collision checking within Pinocchio.

@kozakromch Could open a PR solving this issue? Thanks in advance.

kozakromch, 2021-08-02:
I'd like to open PR. But as far as I can see, file robot_hand.py host only here: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/robot__hand_8py_source.html. 
And I can't find it at this repo. 
If I'm wrong, please, show me where this file is located.

jcarpent, 2021-08-03:
The file is here: https://github.com/stack-of-tasks/pinocchio/blob/2dd7e062479d4adcb69aee2dc00f8b641476b28d/doc/d-practical-exercises/src/robot_hand.py
Thanks in advance for your help and commitment @kozakromch

jcarpent, 2021-08-06:
@kozakromch Any news on this?

closed 2021-08-08 by jcarpent