stack-of-tasks/pinocchio #1484: Error when compiling from source with codegen

plancherb1, 2021-08-03:
I'm compiling with codegen enable by running:
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DPYTHON_EXECUTABLE=/usr/bin/python3 -DBUILD_WITH_AUTODIFF_SUPPORT=ON -DBUILD_WITH_CODEGEN_SUPPORT=ON -DBUILD_WITH_OPENMP_SUPPORT=ON
````
However, when I run ```make -j4``` I've found that both on master and on pinocchio3-preview that the compilation fails with a very non-descriptive error
```
[ 55%] Built target test-cpp-rnea
make[1]: Leaving directory '/home/plancher/Desktop/pinocchio/build'
make: *** [Makefile:144: all] Error 2
```
If I compile without the Autodiff/Codegen/OpenMP things work just fine. Anyone have any suggestions?

Cmake output below:
```
which results in:
```
-- Package version (ROS package.xml): 2.6.3
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Performing Test R-pedantic
-- Performing Test R-pedantic - Success
-- Performing Test R-Wno-long-long
-- Performing Test R-Wno-long-long - Success
-- Performing Test R-Wall
-- Performing Test R-Wall - Success
-- Performing Test R-Wextra
-- Performing Test R-Wextra - Success
-- Performing Test R-Wcast-align
-- Performing Test R-Wcast-align - Success
-- Performing Test R-Wcast-qual
-- Performing Test R-Wcast-qual - Success
-- Performing Test R-Wformat
-- Performing Test R-Wformat - Success
-- Performing Test R-Wwrite-strings
-- Performing Test R-Wwrite-strings - Success
-- Performing Test R-Wconversion
-- Performing Test R-Wconversion - Success
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen missing components: dot
-- Since urdfdom >= 1.0.0, the default C++ standard is C++11. The project is then compiled with C++11 standard.
-- Found cppadcg: /usr/local/include (found suitable version "2.4.3", minimum required is "2.4.1") 
-- Found cppad: /usr/local/lib/libcppad_lib.so (found suitable version "20210731", minimum required is "20180000.0") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem serialization system 
-- The Python bindings of Pinocchio will be compiled along the main library. If you want to disable this feature, please set the option BUILD_PYTHON_INTERFACE to OFF.
-- Found Python3: /usr/bin/python3 (found version "3.8.10") found components: Interpreter Development 
-- PythonLibraryDirs: /usr/lib/x86_64-linux-gnu
-- PythonLibVersionString: 3.8.10
-- Python site lib: lib/python3/dist-packages
-- Found Boost: /usr/include (found version "1.71.0") found components: python38 
-- Boost_PYTHON_LIBRARY: /usr/lib/x86_64-linux-gnu/libboost_python38.so
-- eigenpy FOUND. eigenpy at /opt/openrobots/lib/libeigenpy.so
-- Python compiler: CPython
-- Found Boost: /usr/include (found version "1.71.0") found components: unit_test_framework 
-- Doxygen rendering: using MathJax backend
```

jmirabel, 2021-08-03:
Have you checked the complete make log. With `make -j4`, the error message may be buried a bit. Try with `make -j1` to see if you have something more comprehensive.

nim65s, 2021-08-03:
Hi @plancherb1,

Compiling with `-j4` can lead to bury the actual error message far above the last log line. Could just try with `-j1` again ?

jcarpent, 2021-08-03:
It seems you are using gcc which leads to a certain burden of memory when compiling templated libraries.
I would suggest relying on clang instead. Could you output the complete compilation log?

For instance, on the CI due to some memory limits, we are often reaching the CPU/memory limits and use single-thread compilation.

plancherb1, 2021-08-03:
@jmirabel Ah yes of course -- the first error I hit is:
```
fatal error: pinocchio/bindings/python/multibody/pool/geometry.hpp: No such file or directory
    7 | #include "pinocchio/bindings/python/multibody/pool/geometry.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
I am currently re-running it again with OpenMP turned off on pinnochio3-preview (also realizing that code_gen funcs are not set up for OpenMP parallelism anyway as far as I can tell -- so not an option I need then right now). And things seem to be working fine (at about 80%). So FYI there is something up with the OpenMP parallelism compilation but/and I am fine for now! Thank you!

@jcarpent I'm a CMAKE novice -- how do I tell it to switch to Clang (and maybe that's the error)? I also found in general that running Pinocchio with Clang is much faster than gcc -- of course ironically all the code I write seems to be much faster with gcc.

jcarpent, 2021-08-03:
> I am currently re-running it again with OpenMP turned off on pinnochio3-preview

You might need to erase the CMakeCache.txt

jcarpent, 2021-08-03:
> how do I tell it to switch to Clang (and maybe that's the error)?

As simple as `cmake .. -DCMAKE_CXX_COMPILER=$(which clang++)`

plancherb1, 2021-08-03:
@jcarpent thanks! btw it finished just fine with that gcc run (I had rm -r -f the build folder before trying again) -- I'll try again with OpenMP turned on and using clang and let you know what happens.

jcarpent, 2021-08-03:
Thanks for the quick feedback. Just to let you know that OpenMP is only used when running parallel computations (batch computations of RNEA, ABA, collisions algo). Depending on your application, you might not need it.

jcarpent, 2021-08-03:
I close the issue as it seems solved.
Feel free to reopen it @plancherb1 if you feel so.

closed 2021-08-03 by jcarpent

plancherb1, 2021-08-04:
Yup sounds good! Btw in case helpful for you down the road got that same error with the OpenMP stuff:
```
[ 19%] Building CXX object bindings/python/CMakeFiles/pinocchio_pywrap_default.dir/multibody/pool/expose-pool.cpp.o
cd /home/plancher/Desktop/pinocchio/build/bindings/python && /usr/bin/clang++-12  -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DPINOCCHIO_PYTHON_INTERFACE_WITH_OPENMP -DPINOCCHIO_WITH_URDFDOM -Dpinocchio_pywrap_default_EXPORTS -I/home/plancher/Desktop/pinocchio/build -I/home/plancher/Desktop/pinocchio/build/include -I/home/plancher/Desktop/pinocchio/include -isystem /usr/include/python3.8 -isystem /usr/local/include/eigen3 -isystem /usr/include/eigen3 -isystem /usr/lib/python3/dist-packages/numpy/core/include  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -fPIC   -fopenmp=libomp -Wno-conversion -Wno-comment -DPINOCCHIO_PYTHON_CONTEXT_FILE=\"pinocchio/bindings/python/context/default.hpp\" -DPINOCCHIO_PYTHON_MODULE_NAME=pinocchio_pywrap_default -std=gnu++11 -o CMakeFiles/pinocchio_pywrap_default.dir/multibody/pool/expose-pool.cpp.o -c /home/plancher/Desktop/pinocchio/bindings/python/multibody/pool/expose-pool.cpp
/home/plancher/Desktop/pinocchio/bindings/python/multibody/pool/expose-pool.cpp:7:10: fatal error: 'pinocchio/bindings/python/multibody/pool/geometry.hpp' file not found
#include "pinocchio/bindings/python/multibody/pool/geometry.hpp"
```

jcarpent, 2021-08-04:
I have a similar issue on my side. I will fix it quickly.

jcarpent, 2021-08-04:
@plancherb1 I've provided a fix in #1485. Thanks for the report.