stack-of-tasks/pinocchio #1486: finite difference for computing joint velocity and joint acceleration

timothy233, 2021-08-04:
Hi, I have some simulated trajectories for a robot `(q_1,q_2,...q_T)`, where `q_t` is the joint configuration at timestep t. I want to use finite difference to approximate joint velocity and acceleration.

If the joint configuration is in Euclidean space (all joints are revolute joints, no free floating or spherical joints), I can use second order central finite difference to compute them: `v_t = (q_(t+1) - q_(t-1)) / (2*dt)` and `a_t = (q_(t+1) - 2*q_t + q_(t-1)) / (dt*dt)`.

However, my robot is a humanoid and has free floating joint and spherical joints. Thus the joint configuration is actually in Lie group.  I want to ask if there is any function in pinocchio can compute the finite difference (to approximate first and second derivatives) in the lie group? Now, for joint velocity, I use `v_t =pin.difference(robot, q_(t-1), q_(t+1)) / (2*dt)`, for joint acceleration, I don't know how to compute it appropriately.

jcarpent, 2021-08-04:
The first question is: why do you need finite differences, while Pinocchio provides analytical derivatives of most of the functions?
For a_t on Lie groups, you should think of the operation as follows:
```
a_t = ((q_(t+1) - q_t) - (q_t - q_(t-1))) / (dt*dt)
```
knowing that `v_t` and `a_t` evolves on a vectorial space, contrary to `q_t`.

timothy233, 2021-08-04:
Thanks for your quick reply. My task is to estimate ground reaction forces of a humanoid given trajectory of joint configuration and some other contact information. It involves solving an optimization problem where some constraints and losses require q, v, a as input. That's why I am trying to use finite difference for v,a.

To make sure I understand it correctly, do you mean `a_t = ( pin.difference(robot, q_t, q_(t+1)) - pin.difference(robot, q_(t-1), q_t)  ) / (dt*dt)`?

jcarpent, 2021-08-04:
It would be a first estimate, but the second term is expressed in the tangent of `q_(t-1)`. So, you will need to transport it from `q_(t-1)` to `q_t` in such a way both quantities are expressed in the tangent space at `q_t`.

jcarpent, 2021-08-04:
> To make sure I understand it correctly, do you mean `a_t = ( pin.difference(robot, q_t, q_(t+1)) - pin.difference(robot, q_(t-1), q_t) ) / (dt*dt)`?

Just to make sure that this formula will make the job in your case. My previous comments are for when the operators are defined on the identity.

jcarpent, 2021-08-04:
I will close this issue as it seems to be solved.
Feel free to reopen it @timothy233 if needed.

closed 2021-08-04 by jcarpent

timothy233, 2021-08-04:
> > To make sure I understand it correctly, do you mean `a_t = ( pin.difference(robot, q_t, q_(t+1)) - pin.difference(robot, q_(t-1), q_t) ) / (dt*dt)`?
> 
> Just to make sure that this formula will make the job in your case. My previous comments are for when the operators are defined on the identity.

Thanks again for your quick reply! I will use this equation then.