stack-of-tasks/pinocchio #1491: How to add and remove temporary bodies?

hauptmech, 2021-08-12:
My best idea so far is that I have to fake it by manipulating the inertia of the Joint that I want to attach a mass body to. Is this correct?

It would be nice to mark a Frame for dynamic inclusion/exclusion to the mass parameters of a Model. Are the references to Bodies (nbodies, etc) an unimplemented start on this?

jcarpent, 2021-08-12:
@hauptmech It would be nice to a give a context where you aim at adding or removing a body. Do you have a tiny example?

I will then try to provide a precise answer according to this context.

hauptmech, 2021-08-12:
An example would be a manipulator gripper picking up an object and then releasing it. Though grippers imply a final Joint.

Another example would be attaching or removing (during one session of operation) a mission payload to a Joint in the middle of the tree.

An object has an SE3 relative to a Joint and an Inertia

jcarpent, 2021-08-12:
From an algorithmic point of view, I would say that simply adding or removing the spatial inertia supported by the joint itself is sufficient for your case:
```cpp
model.intertias[joint_id] += object_placement_wrt_joint_frame.act(object_inertia);
```

hauptmech, 2021-08-12:
Yeah, I opted to use a Frame (I need the SE3 of the object during use) and since there is no operator- for inertias, a memo of the starting inertia.