stack-of-tasks/pinocchio #1492: Homogenize GeometryObject and Frame

jmirabel, 2021-08-12:
Hi,

GeometryObject and Frame classes both have a parent joint and parent frame attributes but with different names. I keep forgetting which uses what.

I wonder whether it is a good idea to take the opportunity of version 3 to homogenize this. It's an API change, although it would be easy for users to update their code. Any opinion on this ?

jmirabel, 2021-08-12:
I am fine with any naming (though I would say it is easier for users if only one of Frame or GeometryObject API do change).

I agree there are similar information but there are fields (type, inertia) of Frame which do not apply to GeometryObject and they are conceptually different. I would keep then separated or I would introduce a top class `TreeItem` that has parent joint and frame, name and placement.

jcarpent, 2021-08-12:
OK. `previousFrame` seems fine for me. There is no ambiguity at all. Thanks for proposing the notion of `TreeItem`. Or `ModelItem` because we are not only dealing with trees.

jcarpent, 2021-08-24:
@proyan Could you handle it?

proyan, 2021-12-13:
Fixed by #1570

closed 2021-12-13 by proyan